makerbase-mks/MKS-TFT #211: Z:nan

pcollingwood, 2018-09-24:
I have a tft32 connected to a MKS SBASE (V 1.3) in a Delta printer. Config on both have been setup. 
Tapping the Home button, then home from the menu results in homing, but once I have homed then nothing else works in terms of movement.
What I mean is If I go to the movement menu after homing and tap on any of the movement buttons  nothing happens. No movement, nothing. In the top right corner on this screen is Z:nan. This is there ONLY if I home. If I fire it up and go straight to the movement menu the screen shows Z:00.0. I can move the head all over the place using the buttons and the Z:XX changes accordingly. Wherever it was at startup seems to be X0:Y0:Z0.
If I have not homed and use Pronterface I can cause the extruder to extrude. If I have homed the extruder stepper just makes a very short high pitched buzz and moves fractionally.
Any ideas?

DC3D1, 2018-09-25:
In regards to the high pitched buzz try adjusting acceleration and/or the current to that motor.  My double Z motors made a high pitched sound and was spinning so fast that it skipped. I can't think of anything for the other movement issue right now. Might try reflashing your TFT.

pcollingwood, 2018-09-25:
Thanks for the reply. I have dropped the stepper currents to 0.7 and have no noise now.
The Z:nan issue is resolved. I had set my **alpha_max**, **beta_max** and **gamma_max** to 570mm (printable height of my machine) instead of leaving alpha_max and beta_max at 0. I had assumed that all 3 had to be my max height, but in this instance it's not actually referring to the 3 towers, but rather to X, Y & Z positions at end stops. A bit confusing, but the documentation CLEARLY states that alpha_max and beta_max should be 0. My mistake.

closed 2018-09-25 by pcollingwood