makerbase-mks/MKS-TFT #212: 190 secound pauses  every each layer after connecting MKS DET module

bogaskilop, 2018-09-24:
today i connected  MKS DET module to test power outage resume function.

i started simple g code with calibration cube and i observer that every Z height change  printer is stopping at the end of each line just before Z change for next layer and it's taking  19 seconds.

i uploaded my config and simple gcode 
[mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/2411259/mks_config.txt)

[calibration.txt](https://github.com/makerbase-mks/MKS-TFT/files/2411263/calibration.txt)

power resume function works because i tested it when i figured out that printer is stopping every Z change  even after purge line it stops for that 19 seconds  but why ? 

print is from SD card at mks tft 32

bogaskilop, 2018-10-01:
really no answer  from support ?

makerbase-mks, 2018-10-07:
@bogaskilop Wouldn't this be the problem if you didn't connected the DET module?

mcsultan, 2019-01-06:
if, m105 in your gcode, del all.

closed 2020-06-06 by makerbase-mks