makerbase-mks/MKS-TFT #215: MKS TFT + Baby stepping

selvakumaran2016, 2018-09-28:
Hi All,

I am using MKS Gen 1.4 + MKS TFT + Marlin.

In Marlin, we can use Babystepping. With 2004 LCD, baby stepping is coming easily. But I am unable to do babystepping with MKS TFT.  I tried to add an icon in the 'More' option by using M290 X--, M290 Y--.

But it is not working. Any suggestions please.

Thanks 
Selvakumaran

makerbase-mks, 2018-10-07:
>moreitem_button1_cmd:M290 X--;M290 Y--;

selvakumaran2016, 2018-11-02:
I tried a lot. Its not working. Please suggest any other idea.

makerbase-mks, 2020-02-27:
Please send me your configuration file mks_config.txt

3dworkio, 2020-02-27:
Hello,

I have the same issue, can you confirm me where to add this line?
I attach the file, thanks in advance!

Regards!

[mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/4261099/mks_config.txt)

makerbase-mks, 2020-02-28:
If the screen is idle, you can set it in the configuration file like this:
#set number of "More" button
> moreitem_pic_cnt: 1

#edit command for 1 ~ 7 "More" button, each command must be separated by semicolon ";"
> moreitem_button1_cmd: M290 X-; M290 Y--;

If the screen is printing, you can set it in the configuration file like this:
#how many "More" button display on screen interface
> morefunc_cnt: 1

#command setting on button1 ~ 6
> morefunc1_cmd: M290 X-; M290 Y--;

closed 2020-05-27 by makerbase-mks