makerbase-mks/MKS-TFT #216: No hotend 2 preheat button

chasmac340, 2018-09-28:
I have a tft32 connected to a sbase board, it is set up for dual extruder. Both extruders show up in the extruder window but not in the preheat window what needs to be done?

makerbase-mks, 2018-10-11:
Hello, generally in the case of dual nozzles, the touch screen firmware has a  dual extruder single nozzle setting, the default is enabled, so you share a thermal in the preheating interface, so you can't switch, but the extrusion interface is OK. Switched. As long as you don't enable that feature in the configuration file, then re-flash, you can switch in the warm-up interface (provided you set the dual nozzle).

nsongara22, 2019-07-01:
there no option for extrude from extruder 2