makerbase-mks/MKS-TFT #217: tft70 1.0.1 WiFi unstable

rafaelferrari0, 2018-10-03:
Hi, I'm using TFT70 v1.1 with latest firmware (mkstft70.bin and MksWifi.bin),
The wifi module is an Mks Wifi, already connected, the signal is good.

But connection normally is very unresponsive, always hang my printer.
Tested with Pronterface 1.6.0 (tcp streaming mode = Off), and Repetier-Host (ping-pong = On)

And I can't transfer files using M28 command, the display shows "State: STA    Exception"

makerbase-mks, 2018-10-08:
First, you need to enable the cloud service: > cfg_cloud_enable: 1。You can click the button of cloud service  in the WIFI interface. Can you see a QR code and a serial number?  you can configure WIFI module to STA mode if you need to use MKS_Clound APP to transfer files. You can try to replace the network if the transfer fails. Using the Pronterface software to enter the IP address in the port.for example: 192.xxx.x.xx:8080, you should make sure  computer is on the same network as the WIFI module.

rafaelferrari0, 2018-10-10:
Ok, I've enabled cloud and got a serial number. But I'm connecting via local network, my local IP is 10.0.0.170. I connect with Pronterface (tcp streaming = off) or Repetier-Host  (ping-pong = on) or Simplify3D (using HW Virtual Serial Port) in port 8080 to control the Printer. When using TFT32 using same network, same parameters, the printer always respond to commands and prints. But with TFT70, the printer is unstable, some commands are lost, hangs, stops printing. Using 57600 baud (cfg_baud_rate:2), and MKS SBase, I believe its TFT70 firmware because already tried another MKS WIFI (and uploaded the MksWifi.bin firmware).

MS1987, 2018-10-13:
@rafaelferrari0 , so you mean , using mks wifi+ tft32 + sbase, is quite ok; But using mks wifi + tft70 + sbase, is not stable.

rafaelferrari0, 2018-10-13:
> @rafaelferrari0 , so you mean , using mks wifi+ tft32 + sbase, is quite ok; But using mks wifi + tft70 + sbase, is not stable.

yes