makerbase-mks/MKS-TFT #218: MKS tft28 tft35 no communication

corelc, 2018-10-11:
Hello! I have acquired a tft28v4.0, as soon as I installed it and had no communication with my ramps 1.4, I changed ramps and arduino but the problem persists. I then bought a tft35 v1 to build my printer but the same thing happens! what I noticed is that if I connect the usb port and open the repeater host it indicates the temperature on the screens but does not control anything and the numbers flash every 2 seconds approximately
All the settings are fine, including wifi

MS1987, 2018-10-13:
@corelc , be sure that when using tft28/tft35 to control, cannot use the pc software, Repetier Host eg, to connect the mainboard at the same time.

corelc, 2018-10-13:
Thanks MS1987 for answering, after many tests without connecting usb to the pc, it occurred to me to connect to know if it had connection with the arduino card and it worked! but only indicating the bed and extruder temperatures

makerbase-mks, 2018-10-17:
Hello, generally use ramps1.4, you need to cooperate with tft28 (there is a power interface), because the touch screen is not powered on the motherboard. Regarding the problem of communication failure, it is generally caused by inconsistent baud rate. The suggestion is to confirm the baud rate of Marin firmware. You can also confirm the touch screen and refresh the configuration file (or re-update with the firmware). . When you connect to the host computer, if the screen can display the temperature, the baud rate should be consistent? This is too strange! Do you show the wiring of your motherboard, and what do you mean by photo or video?

corelc, 2018-10-24:
Hello! I just bought an MKS Rumba, I hope this works for me with the tft 35. Anyway, I would like the tft 2.8 to work with the 1.4 ramps for my current printer.
Did the videos come to you? Could you see what may be happening?
Thank you

Peponez, 2018-11-04:
Hello;

 I have the same problem.  I have not communication with my ramps 1.4, I tested the mks tft 28 in other printer but the problem persists. If I connect the usb port without program openned it indicates the temperature on the screens but does not control anything.

Thank you.

winckell, 2018-11-22:
bonjour j'ai le même problème de communication entre ramps et tft28 , quelqu un a t il resolu ce probleme

corelc, 2018-11-22:
Bonjour ! Au moins, je ne pouvais pas résoudre le problème, j’écrivis à MKS et ils répondirent en posant quelques questions de base, après quoi je n’avais ni réponse ni solution. Ensuite, j'ai acheté un tft 32 et il fait exactement le même problème. La conclusion est de dépenser de l'argent de goût, le produit est mauvais!

makerbase-mks, 2018-11-29:
@Peponez 
About the use of tft28 touch screen for ramps1.4&arduino mega2560
. Can you check the temperature when there is a connection to usb, can you move the motor? However, if the usb is not connected, the touch screen cannot communicate with the motherboard.
If you do not connect the USB cable, while providing 12v voltage to ramps1.4, is there any 12v power supply for the arduino mega2560 under the ramps expansion board? (The power supply for ramps1.4 and mega2560 is independent of each other). If the main chip of arduino is quite in a power-off state without connecting usb and 12v, tft28 cannot communicate from the motherboard and obtain temperature.
To print offline, you need 3 12v, tft28, ramps1.4, arduino mege2560.
I don't know if you have three power supplies.

winckell, 2018-11-29:
bonjour, oui je peux lire la temperature sur tft28 avec usb connectée et non je ne peux pas faire tourner les moteurs.
j' ai tft28 et ramps 1.4  alimentés en 12v.
Si je met arduino en 12 v aussi y a t il un risque de court circuit?

winckell, 2018-11-29:
salut, oui j ai branché les 3 sur l alim comme tu  l'as  decris et aucune communication, j'ai commandé la mks v1.4

makerbase-mks, 2018-11-30:
If you have a USB to serial port module (CH340), you can connect the TX\RX pin of the AXU-1 port to check the TFT35 will send/receive data or not.

winckell, 2018-11-30:
je n ai pas de module ch340

550804, 2019-01-06:
Yo tampoco tengo comunicación con la impresora, he cambiado la Ramps, arduino  y la pantalla MKS (las que tengo son las tft 28 V4.0) he modificado los baudios en el marlin así como en la pantalla. También he probado a conectarme  con repetier y las temperaturas se muestran en la pantalla. la impresora, funciona perfectamente tanto desde el ordenador como desde la pantalla Reprap. Tengo la configuración con mega 2560 y Ramps 1.4 y fuente de 12v.

misarybar83, 2019-03-01:
Mám TFT28 v4.0 a desku SKR v1.1 a také nekomunikuje přes AUX1. Rychlost je správná.Jáké je řešení?

VALLAPERDE, 2019-04-07:
Friends, anyone who has a solution to running the TFT28 in RAMPS, many are not getting.
Please can someone show you how it works?

BertrandNadeau, 2019-04-13:
Same problem on my side. My baud rate is set as my MKS Gen L 1.0. I refresh my firmware several time (tft 28), but still not communicates. The bed and nozzle temperatures display on the screen, but the touch screen seems inactive. I touch everywhere on the screen and it seems to be freezed. 
I use Aux1 connection with the motherboard. If no answer, I will ask for a refund.

DavidThijs, 2019-04-30:
I have it working with an MKS GEN L, which is in fact a MEGA 2560. The baudrate must be 256000 and not 115200. Now I'm trying to get it to work on an SKR1.1

Danilkz1, 2019-06-28:
Тоже много боролся с этой проблемой. Первое, отдельное питание + 5в, т.к. кушает очень много, стабилизатор нас arduino греется сильно. Во вторник в прошивку выставлены последовательные порты и rx-tx на контактах d16 и d17. И все работает.

Chelito78, 2020-03-25:
> Тоже много боролся с этой проблемой на rampas 1.4. Первое, отдельное питание + 5в, т.к. кушает очень много, стабилизатор на arduino греется сильно. Во вторых в прошивку выставить serial_port_2 y rx-tx en контактах d16 y d17. И все работает.

Hola Amigo. Soy de Buenos Aires Argentina. Tengo el mismo problema. Podrías ayudarme con algunas imágenes y también con la configuración del firmware? Agradecería mucho tu ayuda. Estoy trabado con este problema y no me deja avanzar.

Kodbee, 2020-05-07:
Good afternoon,
I have the same problem. My MKS TFT28 4.0 does not communicate with Ramps 1.4. I have updated the firmwares of the MKS TFT28 4.0 and the Ramps several times, and I have verified that both are at 250000 bauds, but it is impossible communicate. However, when I connect to the computer, I can actually move the motors with the Repetier Host, but I can´t with the MKS. I suspect the problem may be version 4.0. Does anyone have a solution? Does anyone manage to communicate the Ramps with MKS TFT28 version 4.0?
Regards,
-----------
Buenas tardes,
Tengo el mismo problema. Mi MKS TFT28 4.0 no se comunica con la Ramps 1.4. He actualizado varias veces los firmwares de la MKS TFT28 4.0 y de la Ramps y he comprobado que ambas están a 250000 baudios, pero es imposible. Sin embargo, cuando conecto con el ordenador, sí puedo mover los motores con el Repetier Host. Sospecho que puede ser que el problema sea de la versión 4.0. ¿Alguién tiene alguna solución). ¿A alguien logra comunicar con la Ramps con la versión 4.0?
Un saludo,

makerbase-mks, 2020-05-08:
@Kodbee ,can you send some pictures or videos to show how you connected and how is the phenomenon?
lms228@163.com

Kodbee, 2020-05-09:
Thank you. I sent you the video.

antonrr90, 2020-05-16:
Having The Same Problem. I Have A BTT SKR v1.3 MB, And No Communications With The MKS TFT35 v1.0. Anyone know exactly what Marlin Configuration is suppose to be set in the LCD/Control section of the .h file?

Kodbee, 2020-05-16:
> @Kodbee ,can you send some pictures or videos to show how you connected and how is the phenomenon?
> [lms228@163.com](mailto:lms228@163.com)

I sent you the photos and videos some days ago. Have you seen it? I need some help, please

Kodbee, 2020-05-16:
Good afternoon,
I have the same problem. My MKS TFT28 4.0 does not communicate with Ramps 1.4. I have updated the firmwares of the MKS TFT28 4.0 and Marlin several times, and I have verified that both are at 250000 bauds, but it is impossible communicate. However, when I connect to the computer, I can actually move the motors with the Repetier Host, but I can´t with the MKS. I suspect the problem may be version 4.0. Does anyone have a solution? Does anyone manage to communicate the Ramps with MKS TFT28 version 4.0?
Regards,

MS1987, 2020-05-26:
@Kodbee , sorry for so delay response. I check your files, and just now don't find the reason. But I want to ask : you connect the RAMPS both with the PC and TFT28 at the same time? If so, maybe you should disconnect the PC as they are sharing the same uart.

Kodbee, 2020-05-27:
Thanks. It does not work even if it is disconnected from the PC...

jquintela73, 2020-06-02:
Another with the same failure, I have a mks Sgen L and TFT 3.5 with marlin 2.5.3, but it does not detect the TFT updated to 1.0.6, the printer works perfectly since I print with the PC connected, but the screen does not work, it supplies current but nothing else, you can walk through the menus, but does not recognize the temperature sensors or movements.
Any possible solution?

jquintela73, 2020-06-05:
RESOLVED, change port from -1 to 0 and it works.(#define SERIAL_PORT 0)
now the pc no  recognizes the board.
there is no way to run the TFT with the connected pc?

Kodbee, 2020-06-06:
Thanks, but I had my port defined in that way (#define SERIAL_PORT 0), and don´t works in any way...
Don´t connect with MKS TFT, with or without connect with the PC...

PyroOsky, 2020-06-19:
Hello, same problem here. 

I can move the motors in repetier but nothing happens with the MKS TFT 28.

I tried everything : Change SERIAL_PORT and SERIAL_PORT_2 from -1 to 0 or from 0 to -1 , change baud rate, add external power supply for each board (MKS TFT + Ramps 1.4 + Mega), update MKS TFT and change values, connect or disconnet USB from PC...

Strange fact, with baud rate of 250000 both on MKS TFT and Marlin, I can read the extruder temperature, but not with a baud rate of 115200.

See pics attached for more wiring details (even if it's messy because of all these tests).

Thanks for you help !

Ps : I use a good old Geeetech i3 Pro B Aluminium from which I changed the original motherboard for a Mega 2560 + Ramps 1.4

[image]
[image]

Kodbee, 2020-07-29:
It´s impossible. MKS TFT28 4.0 **DOESN´T WORK** with Ramps 1.4. I am trying communicate both since 3 months ago, but it is impossible. MKS´s people don´t solve this issue. Money waste...

Kodbee, 2020-08-05:
The MKS TFT28 v4.0 display still does not function properly with the ramps. I have noticed that although I don't move the motors when I try to do it through the MKS display, I can still read the extruder temperature. On the other hand, I have noticed that in the HOME section, the icon Motor-Off does not respond, it cannot be changed to Motor-On, I think perhaps the failure may be related to this. Can someone help me?
[image]
[image]

rogeriollorente, 2020-08-07:
Good afternoon, buy an MKS TFT35_FW to connect to my 1.4 Ramps, but I can't do that, can someone help me? MKS TFT35 connects to Ramps 1.4?

Kodbee, 2020-08-07:
Welcome to the club...

rogeriollorente, 2020-08-07:
Hello, my MKS TFT35 does not connect with my Ramps, even changing the Baud rate does not connect, moreover, the screen gradually disappears, I have already checked the connection with ramps several times without success ....

polloloco96, 2020-11-18:
alguien a podido solucionar el fallo de comunicación he hecho de todo y nada

ADR1990, 2021-08-02:
Would this problem have anything to do with the type of serial port module that the mega is using (CH340 vs ATMega16u2).

ChristoVSmit89, 2022-01-10:
> I have a TFT32 v4.0 working with an MKS GEN L, which is in fact a MEGA 2560. The baudrate must be 250000 and not 115200. Now I'm trying to get it to work on an SKR1.1

Hi there, were you able to figure out how to connect to the skr board  as I am struggling to get the mks tft32 v4. 0 working with a btt skr cr6 board