makerbase-mks/MKS-TFT #219: SD ERR message while uploading firmware in TFT35

ksbhatt89, 2018-10-15:
I am getting SD ERR message while uploading firmware.
I am uploading TFT35_V1.0.1 Release file_windows.
See the attached  image for the error I am getting.
[image]

makerbase-mks, 2018-10-15:
This problem is caused by the screen failing to read the file from sd. It is recommended to format the sdcard and re-update the firmware.

ksbhatt89, 2018-10-15:
I already tried it number of times but problems remains  the same.

makerbase-mks, 2018-10-17:
You check the soldering of the sd card slot on the back of the screen, and there is no line sequence of the solder joint. This may be the cause of this

ksbhatt89, 2018-10-17:
What I see is I am able to upload "TFT28_32_v3.0.2 Release file" firmware on mks tft35 but when trying to upload "TFT35_V1.0.1 Release file_windows" firmware it is showing the same error as in image above.

ksbhatt89, 2018-10-17:
It is not successful. Though it is showing some strange icons on screen but at least it is progressing ahead and it doesn't stop anywhere. Whereas while uploading TFT35_V1.0.1 release file it stops at 99%.

makerbase-mks, 2018-10-18:
Is the firmware downloaded on github? You can try it, re-download, change an SD card for update, see if the situation improves.

manotroll, 2018-10-20:
use fat filesystem

makerbase-mks, 2018-10-20:
If you tft35 touch screen update tft28 firmware, can you go to the main page of the screen? But can't display it properly? But there will be no sd err error?

makerbase-mks, 2018-10-20:
I think the possibility of the problem caused by the sd card is relatively large, you can try to replace a sd card to see if it can be solved. I hope that after you try, if you are not resolved, please contact us. We will also check if it is the reason for the screen itself and find a solution as soon as possible.

ksbhatt89, 2018-11-05:
Thank you all for your suggestions. The problem is resolved. The culprit was SD card Adaptor (casing). I changed in twice. But third time when I changed to new one, I got successful upload.

closed 2018-12-29 by makerbase-mks