makerbase-mks/MKS-TFT #220: MKS-TFT32 cnc Firmware?

Donderda, 2018-10-17:
Hi,
is it possible to provide a TFT32 version of the CNC firmware supporting grbl? I‘d really love to use the display with my cnc router 😊.

blikkies, 2018-10-31:
need this too

openbuilds-engineer, 2018-11-07:
Also need this - or source code so as community we can assist in implementing this for you

Andy-27, 2018-11-16:
I am also strongly interested in getting CNC firmware on the TFT28 / 32
Thanks for kindest support

humphreyd, 2018-11-27:
Please add Firmware for the TFT32 to support GRBL
There is much demand for this functionality to your fine device!

Ruisjm1, 2018-11-29:
I have a lot tft32, I don't want buy 2.4 tft, I want use tft32 on Cnc, please provide firmware!

sokolsok, 2018-12-20:
Is it known when will the new firmware for TFT32 be released?

Skywalkerf34, 2019-01-07:
Hi, I also would be grateful if you can release the CNC firmware for THT32

MackMahon, 2019-01-09:
Yes Please add The CNC for the TFT 3.2..... I am almost finished My CNC build, and I purchased the 3.2 and board for exactly this purpose. I have downloaded the CNC .rar file. and updated the board...but its still just a 3D printer set up.... Please add it. Thanks In Advance

bentrizen, 2019-01-20:
I would strongly like to see cnc support in tft32 screen!

hgroover, 2019-02-10:
> Temporarily it is as long as tft24 firmware, because the firmware used by 24 and 32 is not the same. So for the time being, if there is more demand, maybe the TFT32 firmware will be added later.

Do you have a list of what firmware is supported by what hardware? For example I have silkscreened on the board
MKS TFT32L V4.0
and what appears to be a PCB release date 2018-03-11 beneath that.

In this repo you have the 3.2 and previous versions, also 3.5 and 7.0. Are these other board revisions or are they later firmware revisions? It's not clear. I am just looking for the latest version for CNC with the MKS SBase 1.3 using GCode.

Can you add a list to the top of the repo with boards and what firmware they work with? Or add supported_boards.txt to the different directories?

Antiheld86, 2019-02-16:
Yes, please provide CNC firmware for the TFT32!

Donderda, 2019-02-22:
Any Updates on this?

compunited, 2019-02-23:
Need it too.  Almost all new sets are shipped with TFT 32. I thougt it is certain that there is CNC support

compunited, 2019-02-23:
bought 3 MKS SBase with TFT32 in order  to upgrade my lathe, a new build milling machine and the one in our FabLab. Thought it was a nobrainer :\

Focussys, 2019-02-28:
I also have a TFT32 Mk3.0.1.  It would be great to have CNC firmware for it. I am building clone of a chinese CNC, readily available, but with 3d printed parts.It would look awesome with a touch screen controller.

neocogent, 2019-03-05:
I have a TFT32 on my 3d printer and would buy another to use on my new root 3 CNC build. Please do provide support as I really don't want to step back down to the small screen size of TFT24. TFT32 is already small enough. For time being I am using ESP8266 with ESP3D but I think using a TFT32 would be ideal as I know it works well with my printer.

carl1961, 2019-03-08:
I just bought the MKS DLC and the TFT24 to get a PCB CNC running stand alone. so far CNC TFT24 is useless  Home button does not work, there is no icons to move Z axis, No software to help customize or fix what does not work, we need open source file to work with these TFT screens , there very crippled as thing or.

neocogent, 2019-03-09:
@carl1961 
Thanks for that info. I won't try that route then. I'd love to see an open source firmware for these boards as it could be so much better than it is even for a 3D printer usage. It would give us the flexibility to do what we want and add code back to the community. Obviously the Chinese don't see the advantages of having the community code for them. They're too worried about competitors copying their pcbs. There was one [open source project](https://github.com/robotsrulz/MKS-TFT) for these boards but it has been shelved for a while. I haven't looked at it closely but maybe it could be a starting point for an ongoing OS project. I wish I had time to do that. I have some firmware skills but no time.

carl1961, 2019-03-09:
@neocogent look at the Nextion , Looks like it it going to be nice and they have the nextion editor. maybe that LCD will help MKS get to thinking

sevv61, 2019-03-09:
When love to use my MKS tft32 on my laser engraver

neocogent, 2019-03-09:
@carl1961 
Thx. I found info on itead.cc and some sellers on ebay. It looks very promising and reasonably priced. Even see that MarlinKimbra firmware that has some built in support for Nextion. But haven't seen any pre-made CNC or 3D-Printer screen designs/layouts. It sounds like using the editor it should be quick to design these so I'd expect to see some that people have already made for these use cases. I'd buy one if I knew I wouldn't have to start from scratch on that end.

carl1961, 2019-03-09:
@neocogent , you have the TFT32, there is a CNC for it, I also have the TFT32 so I ordered the shield, hopefully, it works  https://www.makerfr.com/en/cnc/rs-cnc/firmwares-rs-cnc/, and the shield https://www.makerfr.com/en/shop/shield-rs-cnc/ it also uses the [Wii Nunchuk](url) https://www.ebay.com/itm/Nunchuck-Wii-Nunchuk-Video-Game-Controller-Remote-For-Wii-Wii-U-Console-US/153000580494?hash=item239f8b958e:m:mN94zipGyjTMc7jNdubbEdQ[](url)

neocogent, 2019-03-09:
@carl1961
Thx. I'm looking at it now. I have the Gen-L which is a Mega2560 board I program with Arduino IDE as well. WIll check if I can use this CNC fw on that. Presumably can. I have a TFT32 on my 3d printer and if I can flash it for CNC then will buy another for that as I like to share parts for redundancy/spares.

dmdementiev, 2019-03-10:
PLS provide CNC firmware for the TFT32!

V1pr, 2019-04-25:
don't forget CNC firmware for TFT35! (or make the firmware open source)

carl1961, 2019-04-26:
V1pr , the MKS-TFT 24 is not useful, unless they do better then it isn't usable.

V1pr, 2019-04-26:
> V1pr , the MKS-TFT 24 is not useful, unless they do better then it isn't usable.

Thank - not that I've planned to spend another penny with MKS -, than the only reasonable option is to get the firmware source for the community.

Donderda, 2019-06-26:
Any news on this? @makerbase-mks

weed2all, 2019-06-28:
I will love to have cnc firmware on my tft32 as well...so much time pass since this post...any news?

sergiofagundes, 2019-07-12:
Me too. Please there are so MANY people asking for tft32 support, it's time to make it happen.

arferdz, 2019-09-09:
Plus one vote on the cnc firmware for the all the tft line so customers can make the choice what fits for their purpose.  And please make it open source so the community can come in and help it develop

weed2all, 2019-09-09:
Biqu seems to be more responsive at their customers!said that it will be the last mks product I buy...it been pass 11month from the start of this thread and mks don't care about this...if none of us buy mks products anymore mks as a company will not exist anymore...so take a bull by the horn MKS!think of what I've just wrote...

sergiofagundes, 2019-09-11:
I've through the MKS TFT32 to the trash bin. I'll never buy this again, it's a useless tft for anyone using a cnc machine, and why in hell I'll buy a TINY 2.4" tft if I can use a large one ?

carl1961, 2019-09-11:
> I've through the MKS TFT32 to the trash bin. I'll never buy this again, it's a useless tft for anyone using a cnc machine, and why in hell I'll buy a TINY 2.4" tft if I can use a large one ?

go get it back out the trash, you can try this guys version CNC controller firmware for that  TFT32 MKS screen.  https://www.makerfr.com/en/cnc/rs-cnc/firmwares-rs-cnc/

CodelCZ, 2019-11-13:
Hi! It would be really nice to have TFT32 for CNC. Please, make it as soon as possible. It can't be so hard.

carl1961, 2019-11-13:
Go get a Bigtreetech TFT they or open source and people are really getting started customized these. you can compare the original source code with a customized source code and see what they're doing and it would be nothing to customize one for CNC.

VegetableKingdom, 2020-01-09:
I got the news that the firmware of TFT32 for cnc will be released soon,
they said that in 2020, will be open source tft series, I think this is good news

makerbase-mks, 2020-01-09:
Our team is continuing to pay attention to this matter. The firmware supporting TFT32 has been developed, and the current state is in the testing stage. The firmware will be released in a few days.

makerbase-mks, 2020-01-13:
https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS_TFT_CNC

makerbase-mks, 2020-01-13:
Hello，Firmware version supporting TFT32 has been released.

weed2all, 2020-01-13:
Awsome...I'll try it in few hours!

selimayaz, 2020-01-13:
Did anyone tried this? I couldn't get it work.

Ruisjm1, 2020-01-13:
Im missing something or this firmware IS FOR LASER???

Thanks a lot, but i thing we ask for CNC not Laser...

Alex-Ian-Hamilton, 2020-01-24:
I installed the MKS_TFT_CNC firmware successfully on my "mks TFT32_L v4.0", it's showing the laser engraver controls, however none seem to work, including basic movement buttons.
When I had the standard 3D printer firmware installed that did move the CNC fine.

Do you have any suggestions makerbase-mks?

compunited, 2020-01-24:
Doesn't work with smoothieware cnc firmware. Sbase board hangs after button press

CodelCZ, 2020-01-24:
I installed it to my TFT32 v4.0, looks it works fine with both, milling machine and laser engraver. It would be better, but it is only my use case. It is chance to make changes after some discussion?

qulet, 2020-08-02:
Mam TFT 32 V4 i płytę MKS DLC V.2 Mam  drawbot 2 osiowy X,Y  . Mam  dwa ograniczniki  hardlimit i odblokowane Bazowanie  W GRBL. Na kablu USB cykl homowania działa dobrze .
Niestety  nie mogą go uruchomić z panelu TFT 32V4  firmware MKS TFT32_CNC V.1.0.1
Proszę  o pomoc w rozwiązaniu tego problemu.

I have a TFT 32 V4 and MKS DLC V.2 board I have an X, Y 2 axis drawbot. I have two hardlimits and Homing W GRBL is unlocked. On the USB cable, the homing cycle works fine.
Unfortunately, they cannot run it from the TFT 32V4 panel of the MKS TFT32_CNC V.1.0.1 firmware
I am asking for help in solving this problem.

jordisala, 2020-09-21:
I HAVE A MKS DLC 2 BOARD AND A TFT3.2 AND THIS HAPPENS TO ME:

1 VLC DLC BOARD WITH TFT 32 WORKS
2 TFT FRIMWARE WORKS IN LASER MODE WORKS
  2.1 THE MOVEMENT OF THE Z AXIS IS NOT
  2.2 THE SCREEN GOES WELL IN OFFLINE MODE, BUT IN ON MODE
    LINE GIVES ERROR WHEN GRBL'S JOB IS MORE THAN 2
   INSTRUCTIONS
3 TFT FRIMWARE IN CNC MODE DOESN'T WORK OR DON'T DIFFER FROM LACER

DOES ANYONE ELSE HAVE THESE PROBLEMS?

makerbase-mks, 2020-09-22:
@qulet You need unlock GRBL.

rlwoodjr, 2020-09-27:
Is the source code available for the MKS-TFT_CNC v1.0.2?  I am investigating offering an offline controller for BobsCNC routers.

I just ordered 2 MKS-tft35 boards.

qulet, 2020-09-27:
Grbl odblokowuje się po wykonaniu  cyklu  bazowania,  a ten wykonuję się tylko  na kablu USB.
Jak to zrobić bez podłączania  kabla USB ?

jordisala, 2020-09-28:
good morning
i have to say your message was not accurate
my screen is 35 ''

alfonso-mon, 2020-12-11:
hello all work fine on my tft35 only for working we need edit first code nc in candle and erase the first line "1 % in queue"

alfonso-mon, 2020-12-11:
example:
original:
%
G21
G90
G00 X0.000 Y0.000 Z0.000
G01 Z-2.5 F500 S10000 M3
G01 X100

after delete:
G21
G90
G00 X0.000 Y0.000 Z0.000
G01 Z-2.5 F500 S10000 M3
G01 X100

alfonso-mon, 2020-12-11:
hello i am using grlb 1.1 v3.4 from aliexpress and mks tft 3.5

mks-viva, 2020-12-11:
You can try to use LaserGrbl and save as nc file. 
The GRBL has Laser mode and CNC mode, if use Laser mode, so, only X Y axis will move, if use CNC mode, X Y Z will move

alfonso-mon, 2020-12-11:
Excuse me i dont have a laser i use only a spindle

mks-viva, 2020-12-12:
So, are you running it in CNC mode?

alfonso-mon, 2020-12-12:
Yes i use for carving

mks-viva, 2020-12-12:
So where did your NC file come from?

[21 more replies]