makerbase-mks/MKS-TFT #221: tft 2.8 communication with the computer

manotroll, 2018-10-20:
Is there any other way to make tft 2.8 communication on the computer without being over wifi?
I can connect a serial converter where would be the WiFi port to be able to control it through the computer?
to use Repetier-Host for example

makerbase-mks, 2018-10-20:
Behind the tft28 screen is a wifi interface. The computer wants to control the motherboard through a USB or wireless network module. The touch screen is not directly connected to the computer.
The setting of the wireless network module refers to the instruction manual of the touch screen of tft28
https://github.com/makerbase-mks/Datasheet

manotroll, 2018-10-20:
plus I do not want to use the wifi I would like to use a direct connection
it does not have any serial port for this or how did I talk about using tx / rx from wifi for this?
I am using firmware 3.0.0 which is what has bra download this and the latest right?