makerbase-mks/MKS-TFT #222: Rotation screen

alec220, 2018-10-24:
I cannot find in the config file  TFT7.0 line 
##Screen rotation 180 °(0:disable,1:enable)
>cfg_screen_overturn_180:1

Is it mean I cannot rotation screen?!!

makerbase-mks, 2018-11-19:
What is version of your firware. You can try it again with below configuration:
cfg_screen_overturn_180:0xEE

alec220, 2018-11-20:
Hi! 
First I used 1.0.1, but after I return to 1.0.0 becouse 1.0.1 some times "freeze" .
I tried to add this line, but without "
0xEE " it doesnt work.
If you did new firmware, I try it, but when I come back home. It will be at the next week - this week I'm in China, GZ.
--
Отправлено из Mail.Ru для Android понедельник,

makerbase-mks, 2018-11-20:
The rotation screen function of TFT70 has been modified. It will be released soon. If you are in a hurry, you can leave your email address and I can send you a copy directly when you release it.

alec220, 2018-11-30:
Hi!
I came back to Russia, and can to test your new firmware.
How I can use this rotation function? I must to add this line to config file? 
I downloaded firmware from GitHub it looks same - config file dont have this line.
You promise to send new version on my e-mail, I didnt recieve anything yet.

>
>What is version of your firware. You can try it again with below configuration:
>cfg_screen_overturn_180:0xEE

Алексей Киселёв
alec220@list.ru
www.alekmaker.ru
+79058782073
whatsapp, viber
Skype alec220
http://alec220.livejournal.com

makerbase-mks, 2018-11-30:
We have sent it.

alec220, 2018-11-30:
Yes, now I recieved its, 謝謝 ))

>
>We have sent it.

Алексей Киселёв
alec220@list.ru
www.alekmaker.ru
+79058782073
whatsapp, viber
Skype alec220
http://alec220.livejournal.com

alec220, 2018-11-30:
Please translate to English the new config file, half lines I cannot to read - wrong simbols...

>
>We have sent it.

Алексей Киселёв
alec220@list.ru
www.alekmaker.ru
+79058782073
whatsapp, viber
Skype alec220
http://alec220.livejournal.com

alec220, 2018-12-06:
Hi!

I tried your new firware.
1. Return display is good, its works, done.
2 U disk doesn't works.  Display sees it if I plug U disk and after swich on power - only this way! 
If I plug U disk when printer in power on, TFT70 or dont sees U disk ( don't any reaction) or immidiantly freeze...

>
>We have sent it.

Алексей Киселёв
alec220@list.ru
www.alekmaker.ru
+79058782073
whatsapp, viber
Skype alec220
http://alec220.livejournal.com

makerbase-mks, 2018-12-07:
We have sent en config file. The U disk work normally need selected "USB" int the TFT "FileSys" interface.If you do this already  and U disk  not work, you can try to replace the U disk.

alec220, 2018-12-28:
Hi.

  I Tested your new firmware  - Display rotated, OK.
U disk - a little bit better - now I can see U disk if it was connected before  power swich on. If I plug it when printer is on - its doesnt see U disk.
And main problem is: If I plugged I disk before printer on, had seen files on the disk, I can start printing. But it usualy stops about 40-60% printng was done - it just freeze, becouse it lost the U disk- nozzle in model, temperature is hight (worked T)  but doesnt move - I can only push stop and cancel printing

>
>We have sent en config file. The U disk work normally need selected "USB" int the TFT "FileSys" interface.If you do this already  and U disk  not work, you can try to replace the U disk.

Алексей Киселёв
alec220@list.ru
www.alekmaker.ru
+79058782073
whatsapp, viber
Skype alec220
http://alec220.livejournal.com

alec220, 2018-12-28:
Now I tried to switch to SD card - it doesnt work absolutly! It sees SD card, sees files, but when I try to start to print - Display is freeze absolutly - I can only unplug power socket!
Never more I'll buy your boards, I think I have to write about my problem in our 3D forums.... To talk all designers of 3d printers, how I tried 3 month to make new printer with your "New super boards"! 

>
>We have sent en config file. The U disk work normally need selected "USB" int the TFT "FileSys" interface.If you do this already  and U disk  not work, you can try to replace the U disk.

Алексей Киселёв
alec220@list.ru
www.alekmaker.ru
+79058782073
whatsapp, viber
Skype alec220
http://alec220.livejournal.com