makerbase-mks/MKS-TFT #225: TFT32 - all working BUT will not print from SD

shearder, 2018-10-29:
Hey guys

I have an issue where i have the TFT32 and Repetier FW. I saw another issue of the same BUT The thread was closed with a link to version 3.0.2. I have tried this version and it seems i cannot print from the SD card.

Everything works. I cam move it i can preheat i can home etc etc BUT i cannot print. I can select the file and when the print screen shows the temperature remain static and  nothing heats up. All i can do is stop the print.

I have doub;le checked BAUD rates and it is correct at 250000 and the correct version of firmware is selected in the config file. All settings are correct BUT just not printing.

Any ideas?

shearder, 2018-10-30:
OK so i figured when exporting to print from SD that to save and NOT select the binary option in the save dialog!!

Filcanna, 2018-10-31:
It seems it's only possible to print if you have Marlin fw on the 3d printer (according to info I have since now). I'll give a try with Smoothieware (I own an MKS Sbase CARD)

shearder, 2018-10-31:
it all works fine with Repetier just that with Repetier you cant use binary to save. But that was an option i just always selected with no real reason. But no need to really select because when the TFT32 tries to send the code to the MKS board id doesn't know what to send i guess OR the code makes no sense in Binary.

makerbase-mks, 2018-11-19:
@shearder  What is version of your Repetier FW and MKS TFT.
You can send the Repetier FW to me and let's check it.

shearder, 2018-11-19:
The tft firmware I am using is one version behind. The latest didn't work. I think 3.02. Same setup I tried as the latest but that didn't work  repetier is the latest version. 
So it's working but as I say its 1 version behind.

makerbase-mks, 2018-11-19:
It is mean that the TFT old version is work but the latest version in can't?

shearder, 2018-11-19:
> It is mean that the TFT old version is work but the latest version in can't?

Yes that's correct. The latest does not work from SD. So one version behind it works fine no issues.

Filcanna, 2018-11-19:
> I have this exact configuration, Smoothieware on MKS-SBase + MKS-TFT32, and it works just fine. I also have another machine with MKS-TFT35 instead and it works just as fine too.
> […](#)

Yes, I tried TFT on Sbase/Smoothieware and it seems to correctly operate MOSFETs for bed, hotend and parts fan; also no error comes out when trying to print a file. Sbase board still not mounted on my delta, so I'm not aware if it really prints..

shearder, 2018-11-19:
When i get home i will give you the exact version of the firmwares for TFT and Repetier.

Another interesting item is sometimes asn error shows on the screen IF i print from USB not SD.

Something like "expecting line 1 and got 10" or something. But it still prints from USB with that error.

I will record it properly if i see it again. 

But now more important for new firmware to work :)

shearder, 2018-11-19:
Repetier v1.03
TFT32: 3.02

makerbase-mks, 2018-11-20:
Yes, some people have reacted to this problem (checksum line number) before, which will affect the printing. We have solved it in the new version.

robustini, 2018-12-07:
@makerbase-mks when you release the new version?

infomario, 2019-01-22:
hy there, same issue here, everything works but the gcode won't start

delwinbest, 2019-07-31:
I have the same problem.
mks_config has:
#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
>cfg_firmware_type:1

Yet, the unit sends M116 to the main board which is not supported by Marlin, so it seems, giving me:

Recv: echo:Active Extruder: 1
Recv: ok
Recv: ok
[...]
Send: M116
Recv: echo:Unknown command: "M116"
Recv: ok
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: ok

Eternally

delwinbest, 2019-07-31:
I only seem to have this issue with E1 on a dual extruder. E0 test print is fine

Filcanna, 2019-07-31:
> I only seem to have this issue with E1 on a dual extruder. E0 test print is fine

Did you select dual extruder on TFT firmware config?

delwinbest, 2019-07-31:
I have. The only thing that has changed in my set up is that I moved to Marlin 2.0.x

Filcanna, 2019-07-31:
Do the usual checks: comms speed, firmware version, etc. I had this problem with Repetier on  MKS Gen-L controller, solved when I swapped it with a Smoothie compatible card (SBase); now board print beautifully from TFT (4.0 version) UART cable and also from USB/Raspberry Pi 3B+

delwinbest, 2019-07-31:
Yeah I prefer trying to find the root cause...and then fixing it :) before switching the board. 
It worked perfectly (MKS Gen L with stock BIBO2 Touch Dual Marlin 1.0.x FW) . I did not have access to the BIBO factory config and config_adv header files so I've used whats in the Marlin example/BIBO repo folder  when updating the MKS Gen L FW to 2.0.x. Its since then I've not been able to print form SD. 

I can see the bed heats up, and the extruder turns on briefly. Then E1 returns to 0 degrees and it waits forever.

I'll attempt to go back to 1.0.x and see if its a version issue. 

Is there anyway I can get access to the actual source files for the MKS TFT micro controller? The files in this repo are pre compiled unless I missed something.

Filcanna, 2019-07-31:
Yep I’ve found them in Github also, seems that all MKS things are slowly going open-source 

Inviato da me !

delwinbest, 2019-07-31:
I've isolated the problem. 

I've tried printing with the second extruder on Marlin 1.0 and Marlin 2.0. No luck, the second extruder just not heat up (and I believe I get those M116 messages). 

If I change to E0 (first extruder), it works fine and prints as expected from SD Card. 
I suggest anyone running into this try printing from the first extruder. 

It would be great to have both extruders work in the same way from MKS TFT SD Card prints.

Filcanna, 2019-08-01:
What slicer are you using on your computer?

delwinbest, 2019-08-01:
Think I solved it. In my case, I had Octoprint attached on USB which shares the same UART as the MKS TFT. Says in the manual not to use the mainboard USB when using TFT. :/ 
The USB connection was interfering with gcodes being sent form the MKS TFT board.

Filcanna, 2019-08-01:
I have a similar setup, Sbase and RPi 3B+ and TFT 2.8'' v4.0, but
monoextruder, working perfectly

divod12, 2019-10-25:
Uncoment Activate_ok on Marlin canfig.adv and works fine.

MarshMalloW0007, 2019-12-22:
I have a Tornado with an MKS Gen-L and an MKS TFT35.
I confirm that if the USB port of the Gen-L is used at the same time as the screen, the screen does not print.

dissocio0, 2020-11-06:
i installed a tft35 v1.0 to a kossel delta linear plus running smothieware, it was set up with a smart display (ender 3 type)... i connected the tft to the base 1.3 through aux1 and same thing ... everything works except for actually starting a print .... nothing heats up and print never starts ... other than that I can control the printer through the touchscreen

ptitpuma, 2021-05-18:
Same trouble here ... Does anyone have a solution ?
I'm using tft35 and e3d board

diegohilarios, 2023-09-27:
#25157

diegohilarios, 2023-09-27:
[https://github.com/MarlinFirmware/Marlin/issues/25157](url)
A solution to this is remove a line on queue.cpp.

diegohilarios, 2023-09-27:
https://github.com/MarlinFirmware/Marlin/issues/25157(url)
A solution to this is remove a line on queue.cpp.