makerbase-mks/MKS-TFT #228: Makerbase MKS DLC V2.0 Marlin firmware

newtonn51000, 2018-11-07:
Hello , 
I recently buy an MKS DLC V2.0 on aliexpress :
https://fr.aliexpress.com/item/HORS-LIGNE-GRBL-CNC-DIY-starter-kit-MKS-DLC-MKS-TFT24-tactile-cran-CNC-laser-machine/32913813397.html?spm=a2g0s.9042311.0.0.27426c37JXbWEo
And i seach the marlin firmware configuration file for this card .
Can you help me ?
Thank you

stef-ladefense, 2018-11-12:
this DLC card is for CNC with GRBL (ATmega328p).
for Marlin, you need a base with an ATmega2560

newtonn51000, 2018-11-12:
Thank you for your help , i but tjis card because in the description firmware was marlin but the seller make an error 
I flash grbl and it work but the screen freeze  after a few seconde

newtonn51000, 2018-11-12:
Désolé pour mon excelent anglais :-)

stef-ladefense, 2018-11-12:
tu l'utilises pour faire quoi cette carte ? une CNC ou une imprimante 3D ?

newtonn51000, 2018-11-12:
Une découpe laser avtuilement j ai un arduino avec un cnc shield dont je suis ravis  j ai commande ce kit car je voulais pouvoir uliser en stadalone sans pc.
Que ce soit base sur marlin ou grbl n est pas trop un problème mais le gros soucis c est qu'avec  la carte sd la gravure démarre et tout ce fige au boit de quelqueq seconde oblige de tout derancher .

makerbase-mks, 2020-03-12:
Hi! MKS TFT24 support CNC and laser mode's firmware has release, you can test it.

closed 2020-03-12 by makerbase-mks

stef-ladefense, 2020-03-12:
great !

the file is not uploaded !!!!

  | MKS_TFT24_CNC_V1.0.1 Release file.zip | Add files via upload | il y a 2 mois
  | MKS_TFT32_CNC_V1.0.1 Release file.zip | Add files via upload | il y a 2 mois
  | README | Update README | il y a 2 mois

makerbase-mks, 2020-03-16:
You can refer to: https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS_TFT_CNC

uwercb, 2020-04-01:
wie kann ich beim Mks DLC V2.0 die Firmware ändern, wird bei meinem Laser als 200 x 200 erkannt, Arbeitsfläche ist aber 350 x 350

stef-ladefense, 2020-04-01:
> wie kann ich beim Mks DLC V2.0 die Firmware ändern, wird bei meinem Laser als 200 x 200 erkannt, Arbeitsfläche ist aber 350 x 350

english please

uwercb, 2020-04-02:
how can I change the firmware of the Mks DLC V2.0, is recognized as 200 x
200 on my laser, but the working area is 350 x 350