makerbase-mks/MKS-TFT #229: Need Bootloader Bin

openbuilds-engineer, 2018-11-08:
I was experimenting with custom firmware development. Can you upload the Bootloader binary for TFT3.2 V4.00 please?  I have an ST Link so I can reflash it.

makerbase-mks, 2018-11-09:
Sorry, because tft is closed source, there is no BootLoader available.

openbuilds-engineer, 2018-11-11:
> what a pity to close the sources when you have thousands of developers who
> are ready to improve and fully adapt your TFT to Marlin or others.

I dont want the Source code. I just want the Binary to flash it back to
repair my TFT!! Very different thing i am asking!

IDontEatSoap, 2018-11-12:
That's really a shame. I bet they'd get 100x times more sales if this was open source.

I'm quite disappointed with the TFT 35, really.... Very limited and buggy functionality...

openbuilds-engineer, 2018-11-12:
> the hardware is very good, but the software is completely shitty!
> MKS would do well to get out of his asshole

Guys, open your own thread to b*tch about the Source code!   This is NOT
what this thread is about and you guys in here is just lessening my chance
of getting the thing I actually need!

@makerbase-mks -  what I want is not the source code at all, you can keep
it entirely secret.  But you know how you release the BIN file for the
firmware?  I just want the BIN for the SD-Upload Bootloader - I wiped my
flash while exploring the board, and just want to use my ST-LINK to BURN
BACK the default BINs (I dont want to see any secrets or source, just want
to flash the BIN).  Its not a secretive file even, I could extract it from
a working board too but I dont have one on hand.  You can upload the BIN
without worry as it doesnt reveal any of your source code, but it does
allow me to repair the unit

rafaelferrari0, 2018-11-14:
Hi @openbuilds-engineer ,
the STM32F107VC in my TFT32 has 256K of Flash and I think the bootloader is present in mkstft28.bin.
Have you tried flashing this entire file with ST-Link then powering with SD Card (with same firmware files)?
Maybe the bootloader starts and complete the flashing procedure...

openbuilds-engineer, 2018-11-15:
```
ser@User-PC MINGW64 ~/Desktop/TFT28_32_v3.0.3 Release file/Firmware/TFT28 32 firmware/Classic
$ "C:\Users\User\Documents\Arduino\hardware\Arduino_STM32-master/tools/win/texane-stlink/st-flash.exe" write mkstft28.bin 0x8000000
Flash page at addr: 0x08038000 erased
112/112 pages written
2018-11-15T15:50:36 INFO src/stlink-usb.c: -- exit_dfu_mode
2018-11-15T15:50:36 INFO src/stlink-common.c: Loading device parameters....
2018-11-15T15:50:36 INFO src/stlink-common.c: Device connected is: F1 Connectivity line device, id 0x10016418
2018-11-15T15:50:36 INFO src/stlink-common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
2018-11-15T15:50:36 INFO src/stlink-common.c: Attempting to write 231200 (0x38720) bytes to stm32 address: 134217728 (0x8000000)
2018-11-15T15:50:39 INFO src/stlink-common.c: Finished erasing 113 pages of 2048 (0x800) bytes
2018-11-15T15:50:39 INFO src/stlink-common.c: Starting Flash write for VL/F0 core id
2018-11-15T15:50:39 INFO src/stlink-common.c: Successfully loaded flash loader in sram
2018-11-15T15:50:52 INFO src/stlink-common.c: Starting verification of write complete
2018-11-15T15:50:57 INFO src/stlink-common.c: Flash written and verified! jolly good!

```

Flashed (With SD inserted) - doesnt power on LCD, files on SD doesnt change - but good idea @rafaelferrari0

openbuilds-engineer, 2018-11-15:
I still think its a seperate bootloader as a failed firmware flash (like accidentally flashing the 3.5tft firmware using the sdbootloader) doesnt corrupt the bootloader

openbuilds-engineer, 2018-11-16:
> Try this bootloader:

(: that one broke it in the first place. It was for a V1 tft

openbuilds-engineer, 2018-12-14:
Well, tossed this crap in the bin and swore to never support MKS again - done. closing issue

closed 2018-12-14 by openbuilds-engineer

fiegener, 2019-02-05:
I have question about the TFT 35. I tried loading 1.02 and its seemed to mess it up. The splash MKS screen does fully load and now it loads with no button images, just rectangles of where buttons should be or just text. I can navigate but no images. How do I clear it and reload with success. And no mater what I do I cant get my own stash to load. I tried reloading 1.01 and still the same its like it cant access button images. It shows them loading when I update.

delwinbest, 2019-09-25:
> (: that one broke it in the first place. It was for a V1 tft
> […](#)

So i take it you never got the v4 boot load or FW binaries?

iz3man, 2020-04-16:
I bricked my tft35 v1.0 by installing a wrong firmware as well, and need to resurrect it.
I d'led the bin file from the dropbox link, as it seems the correct one for v1.
Can someone help me where to connect the stm32 programmer? I can't find a pinout layout.
Thanks

trucutrubl, 2020-09-01:
How do you connect de STLINK2 to the screen? Is there any diagram or how to?

Thanks and regards.

iz3man, 2020-09-01:
[image]