makerbase-mks/MKS-TFT #230: TFT32 and seal

clon1958, 2018-11-11:
Hello, sorry for my english
Bought yours; the TFT32 v4.0 display is done well, nicely, and works
But it does not fulfill its main functions, it does not print either a CD card or USB.
All recommendations implemented
FlashDisk Secure Digital 4 Gb Silicon Power SDHC class10 card
Connection made by shielded wire, minimum length
Screen power from a separate source
The ZRIB v5.2 printer board is improved, USB is disabled, and the TFT32 screen
Connected directly to MEGA 2560 ports
[image]

Repetier-version 0.92.9 printer firmware

mks_config
#########  Printer type setting  ##################################
#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
>cfg_firmware_type:2 
#machine setting (Normal:1; Delta:2)
>cfg_machine_type:1
#baud rate (9600:1; 57600:2; 115200:3; 250000:4)
>cfg_baud_rate:3
#multi-language(enable:1, disable:0)
>cfg_multiple_language:1
#languages setting
#(simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5). This configuration is valid when "cfg_multiple_language" is disabled.
>cfg_language_type:3
#extruder number(one:1; dual:2)
>cfg_sprayer_counter:2
#enable heated bed(yes:1; no: 0)
>cfg_custom_bed_flag:1
#the max target temp of extruder and heated bed
>cfg_max_sprayer_temperature:270
>cfg_max_hotbed_temperature:150
#pause position (-1 is invalid; Z-axis is relative position)
>cfg_XPOS:-1
>cfg_YPOS:-1
>cfg_ZADD:10

TFT28_32_v3.0.3

The display is working, I can control the printer, but when I start printing, everything hangs
And only printer overload
And one more question :
Why can't I turn on the temperature of the second extruder, it is not even displayed,
Although the feed is displayed, but it is trivial
If you look at the forums, not only this one, but also others, the problem with printing is long-standing, even with your printer boards.
You declare that the display works with all the firmware, the boards are certainly different, but the principle of construction is not the same
And what to do? you have closed firmware code

I want to understand

makerbase-mks, 2018-11-12:
Hello
1. First of all, the printing problem of SD card or u disk
To confirm whether the file system of the touch screen is the same as the .sd card you are using now, it is recommended to use a 4x speed card. The transmission speed of 10x speed is too much, and sometimes it is not stable.
2. Start printing, no response? Is the printer overloaded? Is there any prompt for error on the touch screen?
3. The second extrusion head is not displayed? In the case that your profile has been set? Can you switch between the nozzle and the hot bed at the preheating (extrusion) interface?
I hope you can provide more information.
Thanks for your feedback.

clon1958, 2018-11-12:
Hello
Let's start with the third paragraph
The preheat tab displays the E1 extruder and the table changes.

[image]
[image]

On the extrusion tab, switching occurs between E1 and E2, they change, but as E2 without heating, the engine does not turn on, but on E1 it works, and the temperature is output from the first

[image]
[image]

It turns out that I can not turn on the heating E2
C Repetier-Host heating separately regulated

On the first point: does not print u disk with either SD card
The card and disk are formatted in FAT32. at the expense of the card the other day buy, speed of 4x speed
On the second point
When you turn on printing, the following occurs:
Test: the movement of the print head on the plane, without heating and once
[image]
[image]
[image]
[image]
[image]

And that's it, and he will stand there for a long time, but the TFT32 reacts to the touch
if you stop printing, all TFT322 freezes, but responds to touch, but does not send commands (the analyzer can connect) and the TFT32 overload does not help. But the printer, from its panel, can be controlled, and only the overload of everything helps everything will restore
it turns out, management works, and the seal does not go
We'll keep thinking!?

makerbase-mks, 2018-11-13:
I saw the picture provided.
The problem that the extrusion head 2 cannot be heated at the preheating interface may be because the configuration of the tft touch screen is set in two in and out settings, so when heating, only the extrusion head is heated, as long as it is not enabled in the configuration file. With this option, you can heat the second nozzle.

#Whether to support the function of dual extruder single nozzle.(0: disable, 1: enable)
#the extruder number need set to dual.
>cfg_single_nozzle:0

Followed by the problem of printing
As the picture shows, I have a few questions.
1. Do you not heat the extrusion head and hot bed every time you print?
2. The picture shows that the progress bar has reached 99%. Does the whole process have no action? The temperature is also normal temperature? Will the interface stay at 99% of the interface, or will it end automatically?
3. The print time display takes only 30 seconds or one minute. Is it a problem with gcode (can you provide a gcode for use in your interface)?

You can first set the current hot extrusion head and try again.

clon1958, 2018-11-14:
Hello
With temperature control of extras
You do not have independent control, or can this be done using your own menu or buttons?
I have a rule before printing first, heat everything up to working temperatures, look through the filament, and then start printing, I did this with Repetier-Host and then printed from SD card
It’s not convenient to control the printer from the native panel, but the computer is also bad to keep
Well, the temperature issue is closed !!

There is a question with the seal

When you start printing, no matter where the SD card, u disk on the display error 
                                     
[image]

When you cancel printing, and when you issue a command to the printer, one and the same error crashes.
When overloading, TFT32 crashes the floor, but the display responds to the screen change buttons, only the command buttons cause an error
Printer overload, restores TFT32 performance
printer with hovering display, with native panel works
My test, from the native panel works

G1 Z10 
M117 Reveiew         
G1 X210 Y10	
G1 X210 Y40
G1 X20  Y40
G1 X20  Y60
G1 X210  Y60
G1 X210  Y80
G1 X20  Y80
G1 X20  Y100
G1 X210  Y100
G1 X210  Y120
G1 X20  Y120
G1 X20  Y140
G1 X210  Y140
G1 X210  Y160
G1 X20  Y160
G1 X20  Y180
G1 X210  Y180
G1 X210  Y200
G1 X210 Y10	F6000
G1 X210 Y40
G1 X20  Y40
G1 X20  Y60
G1 X210  Y60
G1 X210  Y80
G1 X20  Y80
G1 X20  Y100
G1 X210  Y100
G1 X210  Y120
G1 X20  Y120
G1 X20  Y140
G1 X210  Y140
G1 X210  Y160
G1 X20  Y160
G1 X20  Y180
G1 X210  Y180
G1 X210  Y200
G1 X210 Y10	F6000
G1 X210 Y40
G1 X20  Y40
G1 X20  Y60
G1 X210  Y60
G1 X210  Y80
G1 X20  Y80
G1 X20  Y100
G1 X210  Y100
G1 X210  Y120
G1 X20  Y120
G1 X20  Y140
G1 X210  Y140
G1 X210  Y160
G1 X20  Y160
G1 X20  Y180
G1 X210  Y180
G1 X210  Y200
M117 Level End...

makerbase-mks, 2018-11-19:
@clon1958 
1.Since Repetier FW V_0.92.9 has made some changes that are not compatible with MKS TFT, we recommend using V_0.92.8 Repetier FW.
2.You can modify the source code in Repetier FW V_0.92.9 to be compatible with MKS TFT. Find the gcode.cpp file in the Repetier FW, find the following code, and comment out:
else if(lastLineNumber && Unable to render embedded object: File ((hasM() && M == 117)) { // once line number always line number) not found.
 if(Printer::debugErrors())
{ Com::printErrorFLN(PSTR("Missing linenumber")); } 
requestResend();
 return;
 }

clon1958, 2018-11-25:
Hi
Well it worked

closed 2018-11-25 by clon1958