makerbase-mks/MKS-TFT #231: mks32 v4 with DET after shutdown don´t align Z

mpcolman, 2018-11-15:
hello, I have the module 220 DET installed in MKS32 v4.0, and when the power line cut occurs the MKS goes into pause mode correctly and when the power line returns I can resume the printing correctly, but if it happens that the power line is cut and then  I am without power in the UPS and everything goes off, once the power line returns and I return to resume the print does not pick up in the head inright place, the Z axis goes below the last state and spoils the impression.
I have set the BLTOUCH and everything seems correct in the config could you help me please?
[mks_CONFIG.txt](https://github.com/makerbase-mks/MKS-TFT/files/2585398/mks_CONFIG.txt)

mpcolman, 2018-11-17:
please I need a response for this issue the tft It does not work for me in this way

makerbase-mks, 2018-11-19:
Correct configuration:
   cfg_single_nozzle:0
Whether the touch screen enters the pause state and the machine has raise the height by 10mm (> cfg_ZADD: 10) before the UPS is powered off . If the machine has not been lifted by 10mm UPS had been powered off. Then power on again, recovery printing will appear your problem. , indicating that UPS has insufficient endurance.

mpcolman, 2018-11-19:
thanks for the response but: Idont have in the config.txt this parameter (cfg_single_nozzle) I have this: #extruder number(one:1; dual:2)
>cfg_sprayer_counter:1
maybe I must to add your param? or change to 0 the value??
By the other hand, the pause mode process is made complete before a total sut down, the head park at 10 at your descrition, so I don't know whuy after restart the power the head moves to home and resume more down the Z axes....  thanks for the support!

makerbase-mks, 2018-11-22:
You can send your firmware of main board and TFT to me. I will test it.

makerbase-mks, 2018-11-22:
What is your main board model.
My email:3003455800@qq.com

mpcolman, 2018-11-25:
sorry mail refused, attached here thanks!

TFT32 4.0   FIRMW. 3.03 WITH DET
Main board data....
MKS GEN 1.4 with Bltouch original

attached instaled complete Marlin 1.1.8
[TEVO_BLACK_WIDOW_Marlin_1.1.8_BLT (4).zip](https://github.com/makerbase-mks/MKS-TFT/files/2613539/TEVO_BLACK_WIDOW_Marlin_1.1.8_BLT.4.zip)

mpcolman, 2018-12-05:
HI NO UPDATES?????

makerbase-mks, 2018-12-07:
You can delete the // of the below configuration in the marlin and set it to zero.
#define Z_HOMING_HEIGHT 0

mpcolman, 2018-12-11:
Hi, didn't work.... do the same mess....................

closed 2020-05-27 by makerbase-mks