makerbase-mks/MKS-TFT #232: resuming print too high?

lulu3d, 2018-11-18:
Hi,

When I use the fillament change function, and click resume print after.., it's like the z-axis accelleration is too high and print is resuming about 1.5 cm above the object?!

All is going OK, when I use the Marlin fillament change on my reprap lcd? Print resume is normal and right height?!

How can I solve this?

Thanks.

makerbase-mks, 2018-11-19:
What is version of your tft firmware. Have you try it in old version of the tft?

makerbase-mks, 2018-11-19:
Send me your configuration file.

lulu3d, 2018-11-19:
Hi,

Here are my specs:
MKS-tft24 (version 3.0.3) but older versions (3.0.1) I had the same issue
Firmware: Marlin 1.1.9
printer: Tronxy X5s, Board: MKS base v1.5

Config.txt (see attachment)

p.s; I never got this right from the start I used the tft24 (always resumed printing too far above the print. I never really paid too much attention and used the standard marlin lcd for filament change, as it is working like it should) But I finally wanted to get things right and know what I do wrong, or how I can change to make it work on both screens ;-)

thanks for your help,

[mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/2594373/mks_config.txt)

lulu3d, 2018-11-22:
I have this line commented out in my Marlin, like it is standard. Should I activate this?

lulu3d, 2018-11-22:
Well, this resolved the issue, but I don't think it is the way to go for me. The 'Z_HOMING_HEIGHT' is a safety feature, in Marlin, to not have my bltouch get stuck in the bed clips, while homing!!

During filament change, I have Marlin parking the nozzle in the home position (left front of my bed) so I can easily whipe off oozed plastic, before resuming. 
Using the standard Marlin way (filament change on my lcd) Marlin resumes by moving to the last print position, BEFORE lowering the Z! This way nothing can be damaged by the nozzle.
In the MKS way, the Z is lowered first, before going back to the last printing point, which can damage the printed parts (which it did in my case ;-))

Not a big deal braker, as I know now, I have to use this function on the LCD, and just don't use it on the tft24 ;-)

Thanks for the help.

closed 2021-02-25 by mks-viva