makerbase-mks/MKS-TFT #233: TFT35 Wire

formu25, 2018-11-19:
Hello, I have a problem with the TFT35.
the load unload wire function no longer works
on my mks sbase v1.3
any idea about the problem?

makerbase-mks, 2018-11-20:
Check that the baud rate of the screen and the motherboard are the same.
Set the firmware type of the motherboard in the touch screen, sbase (the smoothie firmware should be selected in the touch screen configuration file)

formu25, 2018-11-20:
Everything and okay 
The other functions work well
Just the son function has a problem
the problem and appeared since the last update

tft35 / firmware 1.0.1
MKS sbase v1.3 / marlin 2.0

makerbase-mks, 2018-11-22:
You mean the old firmware of TFT is normal?  
MKS sbase v1.3 / marlin 2.0??? sbase+marlin2.0??
Our sbase just run in the smoothware.

formu25, 2018-11-22:
Yes, with the old firmwar it works.
hihi ya MKS sbase v1.3+marlin 2.0 :)

dbemowsk, 2019-12-26:
So I have an MKS TFT35 that I would like to try to use with a Fysetc S6 board.  The S6 board does not have an aux1 port, but it does have a UART1 port that provides serial communication.  My question is, can I just connect the +5, GND, RX and TX to the UART port to make it work or is it crucial to have the other 2 data pins from the TFT35 display connected for it to function correctly?  My understanding of the pinout on the 8 pin connector of the TFT35 is this:

 +5 | GND | RXD | TXD
 +5 | GND |  Dxx  | Dyy

I am not sure of the function of the Dxx and Dyy pins on the display.

closed 2020-03-12 by makerbase-mks