makerbase-mks/MKS-TFT #234: TFT : Version number ?!

stef-ladefense, 2018-11-21:
why do you keep the same number of vesions when you make a release?
how do we know the version to change?

for the TFT32, with version 3.0.3, there is a version on October 23, another on October 29, another on November 14 and another on November 19!
and always the same file name!

Ditto for the other TFT24, TFT35, TFT70!

why do like that?
why not put a letter or a number after, that we know that the version has to change?

examples:
TFT28_32_v3.0.3.a Release file.rar
TFT28_32_v3.0.3.b Release file.rar
or
TFT28_32_v3.0.3.1 Release file.rar
TFT28_32_v3.0.3.2 Release file.rar

how do you want us to follow the corrections?
we do not even know clearly what changes!

thank you 

SB

makerbase-mks, 2018-11-22:
We'll improve it. Sometimes we just modified the configuration file.

stef-ladefense, 2018-11-22:
good sound !
thank

closed 2020-03-12 by makerbase-mks