makerbase-mks/MKS-TFT #235: TFT Display 2.8 v3.0.3 firmware for "retro" doesn't work

TheSimplyCode, 2018-11-23:
Whenever I want to download the Firmware 3.0.3 "Retro/mkstft28.bin" the display shows an error. In the upgrade manager of the firmware! With "Classic/mkstft28.bin" it works.

makerbase-mks, 2018-11-29:
the display shows an error?can you show it by the picture?

TheSimplyCode, 2018-11-30:
Here are two pictures of the display, one with the error and one with the display version:
[image]
[image]

I used following configuration of TFT28_32_v3.0.3 Release:
-mks_config_EN.txt renamed to mks_config.txt
-Retro/mkstft28.bin
-MksWifi.bin
-mks_font
-Red style/mks_pic

thanks

sergioe23, 2018-12-05:
I have the same error, it happens with Retro/mkstft28.bin, doesn't matter what style i use.

makerbase-mks, 2018-12-29:
Hello Killerage,
I'm not sure what the issue is but I'll try to help.
To change the configuration of the Robin you need to change the config file when it is in TEXT FILE (.txt) format and then save that to your sd card, put it in the Robin and then turn on power to start the Robin. When the Robin starts it reads the .txt file makes the changes to the firmware settings and then it edits the file extension type so that it does read it again. If you want to changes the settings again then the config file needs to be in the .txt format or at startup the Robin ignores the file.
To set your steps per mm use the Prusa calculator for x, y and z.
Find that here>>>https://www.prusaprinters.org/calculator/
E is calibrated by running some filament through the extruder and measuring how much went through.
I hope this helps but if not repost and we can try again.
Matt

makerbase-mks, 2018-12-29:
The hardware version is too low, the next firmware version will be compatible with this older version of the hardware

closed 2020-03-12 by makerbase-mks