makerbase-mks/MKS-TFT #242: New firmware when?

robustini, 2018-12-18:
Many people, including me, are waiting for a new firmware that solves the various problems reported, when you plan to fix them and release a finally functional version?
Regards,

Marco

Bobblejot, 2018-12-21:
+2 Plus responses to issues would be nice.

robustini, 2018-12-23:
I think the MKS developers are a little bit panicked...

Bobblejot, 2018-12-24:
I think releasing the source and letting us help to improve the firmware would be a good start, it's not like  its something that needs to stay closed source! would also probably improve sales if it was OS?

robustini, 2018-12-24:
Releasing the code would brutally increase display sales.
You don't have to be a genius to understand it.

makerbase-mks, 2018-12-28:
Sorry for it.We will improve the technical support work in this area.The new firmware is already in beta and is scheduled to be released in the last two weeks.

weanything, 2019-01-07:
OK try as I may I cannot get the Pad7 to upload the firmware in the way described - any help appreciated?

stef-ladefense, 2019-01-14:
ils sont inconscients chez MKS  ! 
ils laisses leurs utilisateurs avec les problèmes issus de leur incompétence !

mettez à disposition les sources et les versions beta ! tous le monde gagnerait par cette action

robustini, 2019-01-29:
> 
> 
> Sorry for it.We will improve the technical support work in this area.The new firmware is already in beta and is scheduled to be released in the last two weeks.

The last two weeks have already passed... where is the firmware?

makerbase-mks, 2019-01-30:
https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS-TFT3.5/TFT35_V1.0.2 Release file.rar

robustini, 2019-01-30:
> 
> 
> https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS-TFT3.5/TFT35_V1.0.2 Release file.rar

I mean for TFT32-28!!!

sandog78, 2019-02-05:
Does anyone have a setup with different icons for the TFT35 in a 1.4 board? no examples like other boards to get a different button layout

Andrei-Pozolotin, 2019-03-06:
See #285

makerbase-mks, 2020-03-12:
Hi! We have released a new version of the firmware.

closed 2021-02-24 by mks-viva