makerbase-mks/MKS-TFT #243: MKS WIFI TFT Client Set up

learn3dprintz, 2018-12-19:
Hey, I recently got a MKS TFT 32 and WIFI Module. Tried setting it up and it works fine in the AP mode. I am trying to activate the client mode, so it can be used remotely, but not able to do so. It does catch the wifi network, but shows IP as 0.0.0.0 and remains disconnected. 

Can anyone please share the steps for Client Set up.

marcohaddad, 2018-12-26:
Hello,

I had this problem when I first set the wifi module on the MKS TFT32 too.
I solved it fixing the IP address since it seamed to be some sort of DHCP
incompatibility with my wifi router.
So, my wifi section in the mks_config.txt file is:

#############  WIFI Function  ##########################################

#wifi model(MKS WIFI TFT:0;  MKS HLK-WIFI:1)
>cfg_wifi_type:1

#wifi mode(AP:1; STA:2)
>cfg_wifi_mode:2

#wifi name and password
>cfg_wifi_ap_name:mySID
>cfg_wifi_key_code:myPassword

#0:disable cloud service 1:enable cloud service
>cfg_cloud_enable:0
#cloud service adress
>cfg_wifi_cloud_host:www.baizhongyun.cn
#cloud service port
>cfg_cloud_port:10086

#Dynamic access IP
>cfg_ip_dhcp_flag:1

#IP/mask/gateway
>cfg_ip_address:192.168.3.100
>cfg_ip_mask:255.255.255.0
>cfg_ip_gate:192.168.3.1

#--------------------------------------------------------------------------------

As you see, even with the cfg_ip_dhcp_flag set to 1, it did not get an IP
from my router, so I set an adequate address manually and never tried to
use DHCP again, even after the upgrade to version 3.0.2

Best regards,

kpdelucia, 2019-02-20:
Could you explain  what #dynamic access ip flag does?  1 =enable dynamic but yet when you chose dynamic you then set the fixed adress below.
I dont even have the below section in my config. they are found in the router settings right?
#IP/mask/gateway
cfg_ip_address:192.168.3.100
 cfg_ip_mask:255.255.255.0 
cfg_ip_gate:192.168.3.1

yanard3, 2019-03-15:
I just got one and I am at a loss. If anyone finds any instructions for setting this up it would be greatly appreciated.
Mine dosen't seem to change regardless of what I change in the configuration..
I can see it with my phone and connect to it but that doesn't help much.
I don't see it connecting to my router even though I give it the ssid and pw.
I can turn it on and off at the touch screen.
Also the ip never changes its alway 192.168.4.1

marcohaddad, 2019-03-16:
It sounds like if the configuration file was not renamed back to
mks_config.txt in order to be loaded again...

yanard3, 2019-03-16:
> It sounds like if the configuration file was not renamed back to mks_config.txt in order to be loaded again...
 YUP.. That was the problem, I do remember reading that now :)
Thank you

yanard3, 2019-03-16:
Well I fixed the connecting problem but now I have a loading problem. It is losing connection as the file is being uploaded halfway thru. I am using pronterface to send a gcode file over the wireless network and it appears to be sending ok but then I get disconnected. The printer starts to heat the bed and extruder and random things happen at the printer and pronterface freezes kinda it looks like it ran out of storage as the file was uploading I am not sure where the file is being stored SD card or mks gen l or mks tft28 memory. Debug shows the commands being sent and acknowledged. the file loads fast and then starts to slow then err's out then locks up. 
Does anyone use this and have had success?

kpdelucia, 2019-03-16:
How did you fix the connection problem.  I am an engineer but not a network engineer.  Can you give me a push in the right direction? I was using the cura plug-in but it never connects

yanard3, 2019-03-16:
I renamed the config file to mks_config after changing the wifi setting in it to
#wifi model(MKS WIFI TFT:0;  MKS HLK-WIFI:1)
cfg_wifi_type:0
#wifi mode(AP:1; STA:2)
cfg_wifi_mode:2
#wifi name and password
cfg_wifi_ap_name:mySID enter your router ssid here
cfg_wifi_key_code:myPassword enter the router password here
#0:disable cloud service 1:enable cloud service
cfg_cloud_enable:0
#cloud service adress
cfg_wifi_cloud_host:www.baizhongyun.cn
#cloud service port
cfg_cloud_port:10086
#Dynamic access IP
cfg_ip_dhcp_flag:1
#IP/mask/gateway
cfg_ip_address:192.168.3.100
cfg_ip_mask:255.255.255.0
cfg_ip_gate:192.168.3.1

the rest doesn't seem to what any difference.
pronterface is the program I am using but I am stuck trying to upload files to the printer 
connect by typing the ip address ie 192.168.1.123:8080 at the port box 
hope that helps let me know if you get it working

yanard3, 2019-03-16:
you can also check to see if you can connect by typing the ip address in your browser it will bring up a setup window from the wifi card

kpdelucia, 2019-03-18:
my config file was a little different see below ... seems to be missing some items
how do you know what these settings mean and of course what to set them to?  Especially your with #ip/mask/gateway.  am i supposed to look these up on my router settings page? or is AP mode making a wifi hotspot to connect to so i choose those.  

I think i have corrupted the file .  i don't see any changes taking place when it boots up.  do you think its the ">" sign at the beginning of each line? I read something about notepad changing the encoding of the file if you don't use "save as ansi"

#############  WIFI Function  #########
#wifi mode(AP:1; STA:0)  ap=printer is host? sta=  printer is client?  **what do these mean**
>cfg_wifi_mode:1  

#wifi name and password
>cfg_wifi_ap_name: (mywifiname)
>cfg_wifi_key_code: (my wifipassword)

#0:disable cloud service 1:enable cloud service 0: disable
>cfg_cloud_enable:0

#cloud service adress
#>cfg_wifi_cloud_host:www.baizhongyun.cn

#cloud service port
#>cfg_cloud_port:10086

#Dynamic access IP
>cfg_ip_dhcp_flag:1

kpdelucia, 2019-03-18:
@yanard3 I found a video of a guy showing that you choose pronterface in Cura. There is a plug in for wireless connection in Cura (the video has no audio)
https://youtu.be/DRbsuUqgL9I

File menu  ＞   Preferences
3. "Print window" section  ＞  "Printing window type"  ＞  Select "Pronterface UI"
4. Click Ok
5. File menu  ＞  Print (or Ctrl + P)
6. Wait for Cura to connect to printer
7. Enter G-code commands