makerbase-mks/MKS-TFT #244: MKS gen1.4 + TFT2.8

FranckGib, 2018-12-22:
Hello,
I installed on my CR10S a MKS GEN 1.4 card and a TFT2.8 screen.
From the touch screen I manage to vary the temperature of the nozzle and the bed, the movement of the motors work. but when I print from the SD card or the USB port nothing happens.
My gcode worked on the original motherboard.
Ps: I'm using marlin and the TFT ecrab has been flashing with the latest FW.
where can the problem come from?
Thank you in advance for helping me

FranckGib, 2018-12-24:
Additional information:
The screen works with a marlin 1.1.8 but not with the marlin 1.1.9.

FranckGib, 2018-12-27:
Thank you for this precision. you have to wait for a new version.
: - ((

makerbase-mks, 2018-12-29:
@FranckGib 
@robustini 
Give me your marlin firmware,and I will test it.

LyaminSS, 2018-12-29:
I have the same problems.
MK4 firmware.
4.3.5 - works. https://marlinkimbra.it/configurator/v4_3_5/firmware_configurator.php
all new 4.3.6 (based on Marlin 119) - randomly printing stops.

upload the file **Configuration_Overall.h**

[Configuration_Overall.zip](https://github.com/makerbase-mks/MKS-TFT/files/2716043/Configuration_Overall.zip)

LyaminSS, 2018-12-29:
[image]
if this line is uncommented, then printing does not start.

robustini, 2018-12-29:
> 
> 
> [image]
> if this line is uncommented, then printing does not start.

Is normal because the MK TFT firmware does not handle confirmation of receipt of each line by the board firmware.
The "randomly printing stops" is not the MK4duo firmware or Marlin, but a TFT firmware problem.
I hope that in the next version it will be solved.

robustini, 2018-12-29:
> 
> 
> @FranckGib
> @robustini
> Give me your marlin firmware,and I will test it.

@makerbase-mks , it would be better if you sent us the beta version of the current firmware, so that we give you confirmation if the problems have been solved.

makerbase-mks, 2018-12-29:
You need set the NO_TIMEOUTS to 0:
I can print it as soon as I modify it.
#define NO_TIMEOUTS 0
//#define ADVANCED_OK - disable

robustini, 2018-12-29:
// Bad Serial-connections can miss a received command by sending an 'ok'
// Therefore some clients abort after 30 seconds in a timeout.
// Some other clients start sending commands while receiving a 'wait'.
// This "wait" is only sent when the buffer is empty. 1 second is a good value here.
#define NO_TIMEOUTS 1000 // Milliseconds

// Some clients will have this feature soon. This could make the NO_TIMEOUTS unnecessary.
#define ADVANCED_OK

If your firmware is able to interpret the "wait" state, the timeout can also be left at 1000 (one second).
Same for the "ADVANCED_OK" feature.

robustini, 2019-01-06:
> 
> 
> It is possible to see a predicted date of resolution of the problem. because in the current state of things I am forced to demote my marlin to use my equipment. a shame for an MKS combo.
> thanks in advance

There's a lot of good beta tester around the world, I'm one of those, so you don't have to touch your prints, there are others who can try a beta / alpha firmware, as long as you let me have it.
Please solve this common issue, thanks!