makerbase-mks/MKS-TFT #248: Filament Runout Sensor

forum999, 2018-12-28:
Is there any instructions on how exactly to wire up the filament sensor?

Also what am i looking for in the firmware to activate it as since the latest firmware it doesnt seem to show up and the word 'shielded' is used alot without explenation?

can you please give me clear cut instructions on how to set up the firm ware and wire in a filament sensor please.

makerbase-mks, 2018-12-29:
‘shielded’ is for customers who do not use this feature to disable this feature.
This is the TFT datasheet, with a labeled pin of filament sensor:
https://github.com/makerbase-mks/Datasheet/blob/master/English datasheet/MKS TFT28 32 DataSheet.pdf

forum999, 2018-12-29:
The link you provided gives a 404 error

And i am aware of the pin what i need is proper instructions on how to set it up, what settings to change in the firmware and what wires i need to connect? 

Ie one wire to b1 obviously is not enough? Where do you connect the common etc...

This is just lazy writing proper instructions are needed

rafaelferrari0, 2019-01-30:
@forum999 , about the link, do a copy and paste, or replace spaces with "%20"
(spaces are bad ;)

https://github.com/makerbase-mks/Datasheet/blob/master/English%20datasheet/MKS%20TFT28%2032%20DataSheet.pdf

Andrei-Pozolotin, 2019-03-06:
See #285

makerbase-mks, 2019-03-11:
https://github.com/makerbase-mks/Datasheet/tree/master/English%20datasheet

closed 2020-03-12 by makerbase-mks