makerbase-mks/MKS-TFT #252: 3.5 Version 1.01 No TXT or Icons

AAShepAA, 2019-01-04:
I can install 1.01 and connect at 25000 baud rate, but I have no text or icons although I see them installing. I do see "ReadyPrint" and faint rectangles where the txt or icons should be.  I am on a Tevo Tornado GenL board.

Thanks in advance.

makerbase-mks, 2019-01-04:
there will be some extra files after use the IOS system to extract our files , such as "._bmp_about.bin", and the file name we use is "bmp_about.bin".

yummyburger, 2019-03-09:
my same, i dont see "._bmp_about.bin", in my though

makerbase-mks, 2019-03-11:
The extra files are hidden. Maybe you can set the hidden files to be visible. Delete the extra files and restore them. Follow us engineers to fix this problem

closed 2020-05-27 by makerbase-mks