makerbase-mks/MKS-TFT #254: sin comunicación Tft 28 con la ramps

550804, 2019-01-07:
No puedo controlar la impresora desde la pantalla.  Cambie la Rampas 1.4, el arduino y la pantalla MKS (Las que tengo son las tft 28 V4.0) ha modificado los baudios en el marlin así como en la pantalla. También se ha probado a conectarme con repetier y las temperaturas se muestran en la pantalla. La impresora, funciona perfectamente tanto desde el ordenador como desde la pantalla Reprap. Tengo la configuración con mega 2560 y Rampas 1.4 y fuente de 12v.

550804, 2019-01-09:
Se habrá comunicado, pero desde la TFT28 no se puede controlar ni la temperatura ni los motores.Se les ocurre alguna solución? o tengo que tirar las dos pantallas que compré.

makerbase-mks, 2019-01-10:
Can you detect that TFT28 is damaged? If so, you can contact our sales to solve this problem.
Each time the TFT28 clicks a button, like the  move button, TFT28 will send an instruction, and receive the command returned by the motherboard. The computer can connect with the AUX-1 interface of the TFT28 through the serial port assistant, click the X+ button of the mobile interface, and the computer serial port software can Received instructions, such as G91; G1 X10 F1000; G90; continuously click the X+ button of the mobile interface, the computer software sends "ok\n" to TFT28, and TFT28 receives an ok and sends the next instruction. In this way, it can be detected whether the TFT 28 can normally send and receive instructions.

makerbase-mks, 2019-01-10:
Can you detect that TFT28 is damaged? If so, you can contact our sales to solve this problem.
TFT28 will send an instruction and receive the command returned by the motherboard when clicks a button, like the move button. The PC can connect with the AUX-1 interface of the TFT28 through the serial port assistant, click the X+ button of the move interface, and the computer serial port software can received instructions, like the  G91; G1 X10 F1000; G90. You can click the X+ button of the move interface continuously , TFT28 sends the next instruction if it receives an ok with the PC software send. In this way, it can be detected whether the TFT 28 can normally send and receive instructions.

550804, 2019-01-21:
Aún no he comprobado nada de eso, pero pudiera ser por la versión de Marlin? tengo la1.1.9

Kodbee, 2020-05-07:
Good afternoon,
I have the same problem. My MKS TFT28 4.0 does not communicate with Ramps 1.4. I have updated the firmwares of the MKS TFT28 4.0 and the Ramps several times, and I have verified that both are at 250000 bauds, but it is impossible communicate. However, when I connect to the computer, I can actually move the motors with the Repetier Host, but I can´t with the MKS. I suspect the problem may be version 4.0. Does anyone have a solution? Does anyone manage to communicate the Ramps with MKS TFT28 version 4.0?
Regards,
-----------
Buenas tardes,
Tengo el mismo problema. Mi MKS TFT28 4.0 no se comunica con la Ramps 1.4. He actualizado varias veces los firmwares de la MKS TFT28 4.0 y de la Ramps y he comprobado que ambas están a 250000 baudios, pero es imposible. Sin embargo, cuando conecto con el ordenador, sí puedo mover los motores con el Repetier Host. Sospecho que puede ser que el problema sea de la versión 4.0. ¿Alguién tiene alguna solución). ¿A alguien logra comunicar con la Ramps con la versión 4.0?
Un saludo,

dinoburgoss, 2020-07-27:
Buenas tardes, tengo el mismo problema una mks GEN v 1.4 conectada a una TFT 32 v4.0 he leido que mandan comandos para saber sila placa no esta dañada por X+.  Aguardo instrucciones.

Kodbee, 2020-07-29:
It´s impossible. MKS TFT28 4.0 **DOESN´T WORK** with Ramps 1.4. I am trying communicate both since 3 months ago, but it is impossible. MKS´s people don´t solve this issue. Money waste...

Kodbee, 2020-08-05:
[image]
[image]

The MKS TFT28 v4.0 display still does not function properly with the ramps. I have noticed that although I don't move the motors when I try to do it through the MKS display, I can still read the extruder temperature. On the other hand, I have noticed that in the HOME section, the icon Motor-Off does not respond, it cannot be changed to Motor-On, I think perhaps the failure may be related to this. Can someone help me?