makerbase-mks/MKS-TFT #256: TFT32 no longer access to smoothie

FlailAway, 2019-01-11:
Hi I have an  MKS-SBASE running smoothieware and an MKS TFT32, v3.0.0

This setup has been working great for over a year and like an idiot,,two days back I checked in here and saw updated software for the TFT32. Need I any more? :(

Not thinking clearly, I just installed the new stuff on the SD card without FIRST, making a backup.  DUH!

The new firmware installs and operates just fine with the touch screen, but it is not communicating with the MKS-SBASE.  I have config set up OK, 3 smoothie, 2 delta, 3 115200.

So, I tried getting an older "History" version 3.0 and even 2.1 but  still not talking to the MKS-BASE.

Any thoughts and/or suggestions? Other than going back in time and making a backup before I did this. :)

Thanks

makerbase-mks, 2019-01-12:
You need make sure that the motherboard has the same baud rate as the TFT32.  You need to note that the configuration file is updated after the file name is changed to CONFIG.CUR.

FlailAway, 2019-01-12:
Thanks for the reply, but I stated the baud setting in the first post. I just went back to a full image backup of my laptop from Jan-12, 2018. I installed the 2.1 version I used when I first got the TFT32 and it was working fine until 3.1 upgrade. Now that 2.1 is not working either. As you can see Arduino is happy with the 115200 rate.

[image]

[image]

FlailAway, 2019-01-15:
So that's it? Throw this one in the rubbish or what?

makerbase-mks, 2019-01-16:
Maybe the component with the silkscreen number D9 on the hardware is damaged.
The model number of this component is: SN74LVC1T45DCKT DCK

FlailAway, 2019-01-16:
It was all working fine until I upgraded to the 3.03 firmware. I removed the SD card, deleted the old firmware and installed the new stuff. Plugged the SD card in and it went through the booting and flashing routine. But now nothing from the MKS SBASE + Smoothieware board.

How would I test D9 ("TAF" printed on it)? I have the PDF data-sheet for it but no way to test it.

makerbase-mks, 2019-01-17:
TFT28 will send an instruction and receive the command returned by the motherboard when clicks a button, like the move button. The PC can connect with the AUX-1 interface of the TFT28 through the serial port assistant, click the X+ button of the move interface, and the computer serial port software can received instructions, like the G91; G1 X10 F1000; G90. You can click the X+ button of the move interface continuously , TFT28 sends the next instruction if it receives an ok with the PC software send. In this way, it can be detected whether the TFT 28 can normally send and receive instructions.

makerbase-mks, 2019-01-17:
TFT32 will send an instruction and receive the command returned by the motherboard when clicks a button, like the move button. The PC can connect with the AUX-1 interface of the TFT28 through the serial port assistant, click the X+ button of the move interface, and the computer serial port software can received instructions, like the G91; G1 X10 F1000; G90. You can click the X+ button of the move interface continuously , TFT28 sends the next instruction if it receives an ok with the PC software send. In this way, it can be detected whether the TFT 28 can normally send and receive instructions.

FlailAway, 2019-01-20:
Serial is working as you can see from the Arduino Console in my post above, on booting it sent "smoothie ok ok"

The TFT **receive** from the MKS-SBASE board is not working as you can see from the post above where there are zeros for the temp "0/180" and and the other attached (using EXP1 and EXP2) display is showing 022/000. It IS receiving data from the MKS-SBASE board.

Tapping the "X+" or any other movement buttons on the TFT do nothing, so pretty obvious that the **send** to the MKS-SBASE board is also not working.

mremingtonb, 2019-08-16:
Hey FlailAway, I'm having a similar problem with no communication between the TFT and the board even though the baud rates are the same.  Did you ever get it working?

K16-tech, 2019-11-18:
Faced the same problem.  I decided by setting the maximum communication frequency between the board and the screen, both there and there.

pcollingwood, 2020-10-27:
I had the same problem. After working FINE for a couple of years the TFT32 just stopped communicating with the smoothieboard. I hadn't changed anything on the TFT or smoothie, as I had set both to 115200. Just to test I changed only the smoothie to 250000 and it started working again. I believe that the TFT must have reset itself and changed its speed to 250000. Is there a way to see the TFT settings without just flashing it again? I would be interested in seeing what is set. I have the config file that I originally flashed it with and it says 115200, but that's moot, as something clearly changed.