makerbase-mks/MKS-TFT #259: Feature request: Babystepping

petieken, 2019-01-14:
I have been using the TFT32 a couple of days now, and I really miss the babystepping function that I used on my Reprapdiscount 2004 display. On my delta printer I always use the babystepping to get that perfect first layer, because different kinds of filament (ABS,PLA,PETG,...) need a different height, especially PETG. Babystepping works much faster for me than figuring out that first layer height.

If this feature could be implemented in the option menu when printing, then I could get rid of the reprapdiscount display on my printer.

robustini, 2019-01-15:
Just add a custom buttom with the gcode command "G92 Zx.y" or "G92 Zx.y" for negative, for example "G92 Z0.01".

petieken, 2019-01-15:
Thanks for the reply, I had already tried that with the M290 Z0.01 command and it works. 
The problem is there is no text under the icons, so when the screen is full with icons without text it gets confusing. For me the "MORE" function is not so handy unless there is an option to add your own text.

makerbase-mks, 2019-01-16:
@petieken You can design the icon with text.

petieken, 2019-01-16:
I don't want to design an icon with text in it. I want text below the icon, like the standard icons. Is this possible?

makerbase-mks, 2019-01-18:
You can make a custom design for it.please contact:King@makerbase.com.cn

olimax9999, 2019-02-16:
after a long time searching for a working babystepping, here are my result

>morefunc1_cmd:G91; G1 Z0.1; G92 Z0; G90; M500;
>morefunc2_cmd:G91; G1 Z-0.1; G92 Z0; G90; M500;

it's working on a cr-10 with unified firmware, so it probably work on standard marlin

Andrei-Pozolotin, 2019-03-06:
See #285

sancocbvb, 2019-08-26:
@robustini how to add custom buttons in mks tft28 please tell if you know and which file to edit in firm ware to add custom buttons @petieken @makerbase-mks @Andrei-Pozolotin