makerbase-mks/MKS-TFT #260: No Z moving button in CNC version.

Quajetson, 2019-01-14:
Please add Z+ and Z- in moving screen for CNC firmware.

makerbase-mks, 2019-01-18:
Hello,you can add the function of "More Buttons"  in the settings interface.

targetorsk, 2019-02-26:
give an example of how to do it ?

Quajetson, 2019-02-26:
X0 Y0 Z1 F100
X0 Y0 Z-1 F100

hondapa, 2019-03-11:
Hello where would those parameters placed?
regards

Tomaszm700, 2019-04-28:
@makerbase-mks This is not the way to solve this issue. Z axis required also value 0.1 1mm and 10mm If you try to make it on add buttons with G-commands there is no enouge buttons do do it. Why don't you copy this functionality from 3D printers just with add values of 0.1mm in stead of 5mm?

Quajetson, 2019-04-28:
If U need touch panel for CNC, use Nextion TFT. Full customizable with
serial support and graphic editor.

Tomaszm700, 2019-05-01:
I got nextion as well. This is not solution people buying DLC with tft 24 and become non full functional firmware. More than that there is a space for this, in the direction control screen we have two doubled from main screen buttons. Expectation of the customers is to have fully functional CNC TFT Firmware.

Quajetson, 2019-05-01:
Trudno się nie zgodzić. Zrezygnowałem z tego badziewia.

Tomaszm700, 2019-05-02:
Z tego co wiem Chińczycy mają w głębokim poważaniu co my tu piszemy. Zrobiłem TFT 24 ale to jest leczenie syfa pudrem ( dodałem 4 przyciski definiowalne 3 w dół i jeden w górę, tak by można było dojść frezem do materiału.

hondapa, 2019-05-02:
hello can you put the configuration file, how do you have it? Thank you
regards

makerbase-mks, 2020-06-06:
Hi! New version cnc firmware, you can test it.

makerbase-mks, 2020-06-06:
https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS_TFT_CNC

closed 2020-06-06 by makerbase-mks