makerbase-mks/MKS-TFT #264: MKS-TFT-CNC

Andy-27, 2019-01-24:
Hello,
I connected the 2.4" TFT to my DIY Laser Cutter with Arduino Nano and small GRBL compatible board. Use also GRBL firmware on machine. The display is sharing the serial port with computer connected by USB - one or the other - and work very well. However, the customized buttons does not show the right icons. The picture folder contains the icons for "more function" buttons but should have the "custom"-button icons in. I copied the files "bmp_custom1-4.bin" files from other MKS TFT firmware and it works.
The sub-menu "speed" contains a button "move" which does not have any function. What is it for?
The sub-menu "power" contains a button "power" which does not have any function. What is it for?
When choosing a file in carving menu, I am asked to push either borders / figures / cancel. By pushing borders machine works - except the laser (see below). By pushing figures (as well as after finished the borders) a new picture appears where no exit button exist. Meaning a reset is required.
May it is worth to correct it in next revision.

The laser brings full power, if set to 100%, by pushing the light button (open/close). However, when working with .gcode file from SD, the laser never get 100% power. Even if the setting in codes is ...

M3 S255             I set up GRBL the full spindle to 255
G1 X10 Y10 F480
G1.......
......
M5

Do you have any idea why this is not working? If I run the file from computer with Laser GRBL via USB connection the same .gcode file work very well.

Thanks for any help and guidance

makerbase-mks, 2019-01-28:
By pushing borders machine works - except the laser (see below). By pushing figures (as well as after finished the borders) a new picture appears where no exit button exist. Meaning a reset is required.
May it is worth to correct it in next revision.

What do you mean by clicking the confirm button after completing the border and not entering the engraving interface? Someone may engrave the border again.

makerbase-mks, 2019-01-28:
> By pushing borders machine works - except the laser (see below). By pushing figures (as well as after 
> finished the borders) a new picture appears where no exit button exist. Meaning a reset is required.
> May it is worth to correct it in next revision.

What do you mean by clicking the confirm button after completing the border and not entering the engraving interface? Someone may engrave the border again.

> The laser brings full power, if set to 100%, by pushing the light button (open/close). However, when working with .gcode file from SD, the laser never get 100% power. Even if the setting in codes is ...

Is the power of your engraving different from the power of your gcode?How do you determine that the power of the two is different?

Andy-27, 2019-01-28:
Hello,
here my response:
_What do you mean by clicking the confirm button after completing the border and not entering the engraving interface? Someone may engrave the border again._

The file Pic01.jpg shows the screen after choosing a file to be printed. I have created a small g-code file just making a square of 10mm to be cut.

The file Pic02.jpg shows the screen after machine has executed the g-code. What is also visible here is that the text has no wrap. It is just one line which is not readable.

The file Pic03.jpg shows now what happen after pushing "confirm" from Pic02 or "Figures" from Pic01. On this screen is no button to escape. Only "reset" help to step out.

Furthermore, I am missing a "Homing" button e.g. G-Code G28. There is only a home button where the machine runs to the coordinates which has been set as "0".

Thanks a lot for help

[image]
[image]
[image]

Andy-27, 2019-01-28:
In regards to the 2nd question:
When running the small g-code file from computer with Laser GRBL, the Laser power is 100%. In GRBL I have chosen as spindle max. 255 and in g-code file I set to 100% with M3 S255.
On a sheet of thicker paper the square of 10mm is cut off.
By using the same g-code file on SD card and run from MKS-TFT, the power is around 10%. The machine is running the square but the laser is not even burning slightly the surface - you see nothing on paper. In MKS-TFT menu "Power" it is set to 100%.
What I also found out, the speed I choose in g-code file with F480 and Laser GRBL runs the machine with that speed. The g-code file from SD card in MKS-TFT is ignoring the speed. It runs much faster and has no influence on what is selected in "Speed" menu, 100% or 10% makes no difference in machine run. This selection only applies to manual move from "Move" menu.

I have added my test file - just changed extension from .nc to .txt

Thanks for help
[Test 3.txt](https://github.com/makerbase-mks/MKS-TFT/files/2803353/Test.3.txt)

makerbase-mks, 2019-01-31:
https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS_TFT_CNC/MKS_TFT_CNC_V1.0.0 Release file.rar

closed 2019-02-20 by makerbase-mks

qulet, 2020-11-17:
Proszę o pomoc!
Mój zestaw to:
MKS DLC 2.0  + MKS tft32_CNC V1.0.2
System się zawiesza przy wykonywaniu polecenia "BORDERS" polecenie "FIGURES" działa "OK" Moja maszyna ma  przełączniki krańcowe w osiach X,Y i Z.  Polecenie "Home" działa z "Custom Button" .Dlaczego nie działa polecenie "Borders" ?

qulet, 2020-11-17:
mój mks_config.txt

[
[mks_config_EN.txt](https://github.com/makerbase-mks/MKS-TFT/files/5555579/mks_config_EN.txt)
](url)

tuxcnc, 2021-08-06:
Hi.
I have bought TFT3.5 and I am surprised.
What are "BORDERS" and "FIGURES" ???
I write g-code and I want run it.
I DON'T WANT any "BORDERS" and "FIGURES", I want "RUN" code only.
How can I remove this stupid submenu?
Regards.

wiring80, 2021-11-07:
@tuxcnc: ++thumbs up
I dont understand it too. Seems to make non sense for me, the gcode tell all what i want from that maschine.

In my case, i use my MKS DLC with an TFT24 as a plotter and a laser engraver. If i put gcode to CD Card and feed the TFT24 with it, it doesnt start. 

I have two gcodes on sdcard saved. One to home cycle and set the pen in the right height, and one to plott an smiley. The first one starts right, but my smiley doesnt do it. From OpenBuilds Control streamed to GRBL seems to be fine. 

Im really disappointed over that piece of hardware, with fully uncompleted code/firmware that even not provide simple basic functionalities. And furthermore bads, it isnt opensource