makerbase-mks/MKS-TFT #267: release all firmwares

stef-ladefense, 2019-01-29:
let the community participate in the debug and feature additions

rafaelferrari0, 2019-01-30:
Yes, indeed. Doing this will help us to improve functionality and fix bugs (lots of). Makerbase will benefit, there is a chance to grow up this community, spread the product with new functionality, ideas and uses.

Operrr, 2019-01-31:
I fully support the idea

kkaleta, 2019-01-31:
Giving us just a 'More' menu, it's not enough.

Operrr, 2019-02-02:
I don`t understand

>
>Giving us just a 'More' menu, it's not enough.

sandog78, 2019-02-07:
I SAY DO IT!!!
Give the people what the want..

stef-ladefense, 2019-02-18:
and as usual MKS makes the dead head in the sand ...

kkaleta, 2019-02-20:
I bought MKS-TFT 32 and after 2 weeks selling it. Because i cant change my
z-probe offset, which is crucial to me. This software should be opened.

rafaelferrari0, 2019-02-21:
Yes, this is why the firmware must be opened. There are so many changes, ideas, bugs to be fixed.
And if MKS is not able to fix software problems and improve new ideas, we are !

> I bought MKS-TFT 32 and after 2 weeks selling it. Because i cant change my z-probe offset, which is crucial to me. This software should be opened. 
> -- *Kamil Kaleta*

iz3man, 2019-03-01:
You can't change z-probe offset on the fly? This is a must have for all printers.

Andrei-Pozolotin, 2019-03-06:
See #285

stef-ladefense, 2019-04-21:
MKS fucked us with his screens and shitty softwares!

iz3man, 2019-04-21:
I already ditched it and use a 'regular' 12864 display. Does all it needs to do and no need to modify.

darkspr1te, 2020-01-31:
There is a open-source tft firmware that is compatible with the MKS 35 v4, support for others could be added. Theres also a bootloader without that annoying beep and a little colour and choice of bootlogo. So hardware is fine, software you have a choice :-)

stef-ladefense, 2020-01-31:
@darkspr1te 
yes I saw on your github, but I did not find an installation doc which details all the steps to install your firmware on it

darkspr1te, 2020-02-02:
it's all in the README.md 
tl;dr, 
download git zip, extract zip , open extracted folder with platformio , compile, copy mks_tft_1_4_1_2.bin from SDCARD folder to sdcard, rename mkstft28.bin , insert sdcard into display, reset display.

darkspr1te, 2020-02-06:
see https://github.com/darkspr1te/MKSTFT_Marlin_Touch/issues/4#issuecomment-583003260 for info on new features, wifi support with any esp,2 new bootloaders plus debrick guide, precompiled bins/hex files.

darkspr1te, 2020-02-06:
Bootloader for mks https://github.com/darkspr1te/mksbootloader
bootloader for stm32f107/105 in general https://github.com/darkspr1te/stm32f107vc_sd_bootloader, debrick guide for MKS tft 32/v4 is under mksbootloader repo, hex files also

iz3man, 2020-02-07:
> Bootloader for mks https://github.com/darkspr1te/mksbootloader
> bootloader for stm32f107/105 in general https://github.com/darkspr1te/stm32f107vc_sd_bootloader, debrick guide for MKS tft 32/v4 is under mksbootloader repo, hex files also

Does that work for tft35 v1 as well. I'm really confused by all those different versions. 
Thanks

darkspr1te, 2020-02-17:
it might not out of the box but does not take much to port to other devices. the sd-flash part will however work, but it might not correctly initialize your lcd,  we have seen some versions (tft35 v3) that does not do LCD_init in the bootloader but in v4 it does. there is firmware to backup your bootloader , i'am not sure if i've posted mine yet, but i can if needed.

mks-viva, 2021-02-24:
MKS TFT24 : https://github.com/makerbase-mks/MKS-TFT24-Firmware
MKS TFT28/32 : https://github.com/makerbase-mks/MKS-TFT28-32-Firmware
MKS TFT35 : https://github.com/makerbase-mks/MKS-TFT35-Firmware
MKS TFT70 : https://github.com/makerbase-mks/MKS-TFT70-Firmware
MKS TFT Hardware : https://github.com/makerbase-mks/MKS-TFT-Hardware
Thank you for your support MKS's product.

closed 2021-02-24 by mks-viva

appliedmike, 2021-02-24:
Thank you MKS, any chance of the firmware for the CNC TFT35 ? or is the MKS TFT35 applicable for CNC, Laser and Printer?