makerbase-mks/MKS-TFT #273: MKS TFT32_L v4.0 has no effect on board

BorgMaster, 2019-02-07:
The menus work but seem to do nothing, I try to move X direction and nothing, 
I connected the old board to test it and everything works fine as before. 

I bought this TFT and "Board MKS SBASE V1.3 32bit Controller" as an upgrade and I can't get anything to work from this board or touch display. 

Any ideas ?

Thanks !

makerbase-mks, 2019-02-15:
Can it be used normally if the computer's host computer is connected to the motherboard? Is the baud rate of the motherboard set to the same as the baud rate of the TFT?

Oxika95, 2019-03-13:
Hi there I have the same model and the same issue. I am using it to control a Cr10S v2.1 board (baud rate of 115200 I believe) and have thus set the configuration to 3 (which corresponds with the stated baud rate). I have reflashed the board a few times and the issue persists. Using the same baud rate with OctoPrint I am able to send commands that the printer responds to. I am not sure why the screen isn't working.
[mks_config_EN.txt](https://github.com/makerbase-mks/MKS-TFT/files/2961906/mks_config_EN.txt)

Stady234, 2019-04-07:
From everything i have read.. the MKS Touch screen is only compatible with MKS boards.. so it will not work with the cr10s (creality) board...

makerbase-mks, 2019-04-07:
Tft is communicated through the serial port, as long as the baud rate is set
If other motherboards only need to correspond to the tft pin, they can communicate. If it does not correspond, of course, it is not connected.

WeiMingYu, 2019-04-26:
Does the corresponding foot and firmware need to be modified?
Or as long as the follow-up version is connected in the correct position?

closed 2020-03-12 by makerbase-mks