makerbase-mks/MKS-TFT #275: MKS WIFI and ESP3D-WEBUI

GenaUser, 2019-02-08:
Hello,
I use MKS RUMBA board, and I would like to use my MKS TFT35 with the great [ESP3D-WEBUI](https://github.com/luc-github/ESP3D-WEBUI) web solution.
Problems:
ESP3D-WEBUI or web pages can't be uploaded and used in an MKS Wifi modules.
TheTFT MKS screens and ESP8266 wifi module both use the AUX-1 connector (TX / RX pins)
If I connect an ESP8266 module to the TX / RX wifi module interface on the TFT screen, ESP3D-WEBUI doesn't communicate with the board.
If I connect an ESP8266 module to the TX / RX in the AUX-1 on the board (with the TFT screen), the screen doesn't communicate with the board.
Do you have a solution? The coexistence between a TFT MKS and WEBUI screen is possible. It works on a Wemos Mega / wifi card with Ramps 1.4.
If no solution exists, a future update hardware or software is it possible to add this function to the MKS ecosystem?
I don't want to choose between the MKS TFT35 screen and the ESP3D-WEBUI ability for my future printers

Thanks in advance

rafaelferrari0, 2019-02-21:
Hi @GenaUser , maybe OctoPrint ? So you connect it to USB port on MKS Rumba.
But now you need to connect your slicer via network instead of usb cable.

luc-github, 2019-05-30:
@GenaUser I have sent email to @makerbase-mks to get help about communication protocol between TFT and MKS_WIFI_module so ESP3D FW on TFT wifi module can properly communicate with TFT - I am still waitng for answer, I hope to get feedback on this, check https://github.com/luc-github/ESP3D/issues/249
I have just asked how it communicate so they do not need to change anything on their side.