makerbase-mks/MKS-TFT #277: For developers. VERY IMPORTANT

Operrr, 2019-02-11:
I have a sbase and mks tft 3.5 "everything works more or less well, but when I print via USB cable, the display is useless because it does not have a page displaying the current mode of operation. It would be very convenient for the page to be displayed with the temperature of hot-houses and heated platform, something like the “Print” page, then when printing from a computer, the display would have at least some useful function.

In general, a platform with potential and if you make the code open, your product will play with new colors and become more interesting for purchase. Think about it.

sandog78, 2019-02-15:
Is there a way to change the UI on the TFT35.. there are not any examples like in the TFT28 folder

Operrr, 2019-02-19:
I make instruction. see here https://www.youtube.com/watch?v=KjP1nXxx2c0&t=54s

rafaelferrari0, 2019-02-21:
I also need this feature, this can be done if MKS release the source code.

> It would be very convenient for the page to be displayed with the temperature of hotend and heated platform, something like the “Print” page, then when printing from a computer, the display would have at least some useful function.

Good video

> I make instruction. see here https://www.youtube.com/watch?v=KjP1nXxx2c0&t=54s

Andrei-Pozolotin, 2019-03-06:
See #285