makerbase-mks/MKS-TFT #279: G29 stuck on "deploy probe" confirmation

Stady234, 2019-02-17:
To protect the nozzle when performing functions needin the probe, I enabled in Marlin 2.0 the "deploy/sto probe" operation, that simply need the user to click a button on the original screen (dial, rotary, 5button);

This has worked flawlessly, and still works when i reconnect the old screen.  I was hoping that the touch screen would be able to handle this completely...

There are no changes to the touch screen(TFT 32 v4.0) when i run the G29.. I hear the speaker for me to click, but Printer will not do anything unless a button is pushed.

Stady234, 2019-04-07:
Almost two months... no update?

makerbase-mks, 2019-04-07:
Similar situations, we will feedback with the R&D department.
But with some of the new features of marlin, we may not be able to take care of all of them.
So some if you can do it with a custom button

Stady234, 2019-04-08:
Again... due to the serial connection, your screen is not able to send the response the firmware is looking for.

makerbase-mks, 2019-04-08:
Is the screen and usb used at the same time, it will cause serial port conflict

Stady234, 2019-07-24:
> Is the screen and usb used at the same time, it will cause serial port conflict

No... I have the TFT screen and the factory LCD screen connected due to the TFT no being able to send an "enter/ok" to acknowledge the probe being deployed or stowed.

closed 2020-03-12 by makerbase-mks

Stady234, 2020-03-13:
Was this closed for housekeeping or did anything get fixed on this issue?