makerbase-mks/MKS-TFT #285: USF - User Supported Firmware

Andrei-Pozolotin, 2019-03-06:
MKS 很好地复制其他硬件，发明原始软件 - 而不是那么多。
MKS is good to copy others hardware, invent original software - not so much.

Would you rather support your own development, such as:
https://github.com/robotsrulz/MKS-TFT
https://www.patreon.com/robotsrulz/creators

With help of some 27+ people who can already build it:
https://github.com/robotsrulz/MKS-TFT/network/members

Vote here if you are willing to contribute time or money to **USF - User Supported Firmware**.

Andrei-Pozolotin, 2019-03-06:
Here is a letter that I (Andrei) sent to the developer (Roman):
```
Roman, hello.

1. This is a request to revive your MKS-TFT ALT firmware project,
coming from a prospective user.

2. It looks like the critical mass of people to support you have finally arrived:
https://github.com/robotsrulz/MKS-TFT/network/members

3. Please report here if you are willing to resume your work, and what is minimum 
monthly contribution from the community you require. 
Or if you are not willing and why.
https://github.com/makerbase-mks/MKS-TFT/issues/285

Thank you, 

Andrei.
```

robotsrulz, 2019-03-06:
> Here is a letter that user (Andrei) sent to the developer (Roman):

Hi Andrei, let me respond here and outline my position. I am not MKS employee and do not have access to recent MKS hardware. MKS is changing hardware revisions, adding/removing features, etc. To catch up with that I have to spend money on buying all the revisions of MKS hardware to make sure my firmware works on all that updated boards. Also MKS turned on readout protection on recent MKS TFT revisions so if you want to play with the hardware you may end up being unable to restore factory firmware.

All that means that hacking MKS hardware costs money. And my goal is not wasting money but earn some money. And I don't see how I can earn some money with that. Patreon doesn't work in that particular case. So I sold all my stuff and suspended my work on this project.

Andrei-Pozolotin, 2019-03-06:
Roman, hello again.

1. Thank you for the answer, and for straight and simple approach.

2. It sounds like MKS-TFT is a dead-end and we all (1000-s of users) 
wasted time and money on a useless toy and should look elsewhere for replacement.
Nonetheless, let me try one more time with you :-) 

3. The intent would be for you to come back to revive the project and then to give it back
to the community so people can support and maintain the project after your initial launch.

4. The acceptable "hand-over level" would be perhaps to support a couple of MKS-TFT boards
and convert build system into something similar to Marlin: A) Atom+PlatformIO for interactive development and B)  some gcc tool-chain for GitHub+Travis automatic builds.

5. Now, I am sure that some users will donate some MSK-TFT boards to you, 
so you get back your hardware library.

6. A "readout protection" and inability to use original MKS-TFT code is a good feature. 
I think most users would rather never look back on that poorly designed buggy stuff again.

7. And, about making money. Say, we, "the community", find among ourselves 
some 200 people, who agree to pay you via patreon $1 / month for one year.
Is it sufficient for your to bring the project to the "hand-over level"?

Finally, it is feasible that this project and MKS-TFT hardware will serve only as a transitional
spring board for creation of crowd-funded open-hardware TFT panel for 3D printers, 
that will run on you original code.

Thank you.

robotsrulz, 2019-03-07:
Hi Andrei!

It's relatively easy to add Makefile to my firmware so it can be built with a generic arm-gcc toolchain and you can use Atom/VSCode/whatever to edit it. PlatformIO however is a no-go, there is no use for it. I can also clone MKS-TFT hardware relatively easy, there is nothing special there, all parts can be sourced easily.

The only concern for me is money at the moment. I had Patreon account for more than a year and earned ten bucks or so in total. Now you are telling me that 200+ ppl wants to support the development. I'm very skeptical about it. I'm getting much more from other projects so unless I see more people there I don't want to put much effort into it.

Best regards,
Roman

robotsrulz, 2019-03-07:
PS: just to show that I know what to do here are some old videos of mks tft with custom firmware printing to marlin emulator and a quick menu overview (in Russian):

https://www.youtube.com/watch?v=LfauV0CJYfo
https://www.youtube.com/watch?v=JpzGEI-_K3Y

Andrei-Pozolotin, 2019-03-07:
Roman:
Surely, "skeptical" is good. 
How many interested users do you want to see to commit to this work?
Andrei.

robotsrulz, 2019-03-08:
Andrei,

At least couple of dozens. I see only you at the moment. Hope it will change soon. Or I will consider this work meaningless.

Best regards,
Roman

Andrei-Pozolotin, 2019-03-10:
Roman:

1. OK, "couple of dozens" is fine. Lets make it 50. Let me see if I can find 50 people in the next month or so. If this community can not produce 50 paying members, then these people deserve the firmware they already have. :-)

2. Can you please un-archive the [project](https://github.com/robotsrulz/MKS-TFT)? This would indicate a "tentative resurrection", and allow to have a place to continue conversation, etc.

3. Can you also activate [gitter](https://gitter.im) for the project? That would help with real-time "thought exchange". BTW, what is current time zone?

4. I would like to send a PR to the project with `readme.md`, explaining "the plan", "rules of the game", etc., if that is ok with you.

5. Can you please setup [Patreon](https://www.patreon.com/robotsrulz/creators) specific for this project only, and in such a way that people can register, but delay actually charging them till the point in time when subscription level actually hit the 50 supporter threshold?

Andrei.

3kmch, 2019-03-15:
I'll throw in $20 as soon the project starts. :)

Andrei-Pozolotin, 2019-03-15:
@3kmch Kurt: thank you :-) 
Please spread the news about this call for support in whatever forums you have a voice.

FranckGib, 2019-03-23:
I am not a developer but I follow you! congratulations to relaunch the subject.
I am also frustrated by the lack of responsiveness of MKS debugging his screen.
Congratulations for your project!

Projectile007, 2019-04-03:
I'm in for another $20

Tomaszm700, 2019-04-28:
We are both in, Me and Friend of my. I sent you email today with info that we can help with graphics and development. You can expect 2x20$ monthly from us + our offered help.

atolala, 2019-05-05:
https://github.com/atolala/MKS-TFT32-CNC

darkspr1te, 2020-01-01:
> An old Chinese proverb says:
> 
> ```
> MKS 很好地复制其他硬件，发明原始软件 - 而不是那么多。
> MKS is good to copy others hardware, invent original software - not so much.
> ```
> 
> Would you rather support your own development, such as:
> https://github.com/robotsrulz/MKS-TFT
> https://www.patreon.com/robotsrulz/creators
> 
> With help of some 27+ people who can already build it:
> https://github.com/robotsrulz/MKS-TFT/network/members
> 
> Vote here **+1** if you are willing to contribute time or money to **USF - User Supported Firmware**.
> If possible, please mention how much monney can you contribute per month via Patreon to the developer.
> 
> And drop a line to the developer `Roman Stepanov <robots.rulz@gmail.com>` asking him to pretty please resume his work.
> 
> And support this project on `hackaday`
> https://hackaday.io/project/18852-alternative-firmware-for-mks-tft32

While he has not continued his work , We have been working in the background, [Marlin Touch Port by darkspr1te](https://github.com/darkspr1te/MKSTFT_Marlin_Touch)

Andrei-Pozolotin, 2020-03-05:
@darkspr1te Alex:

[BIGTREETECH TFT35 V3.0](https://www.amazon.com/gp/product/B07VWF4W3J) is still much better, especially with these hacks:
* [user managed config.ini](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/360)
* [remote firmware deploy](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/361)

please make sure to port this back to MKS-TFT :-)

darkspr1te, 2020-03-09:
> @darkspr1te Alex:
>
> [BIGTREETECH TFT35 V3.0](https://www.amazon.com/gp/product/B07VWF4W3J) is still much better, especially with these hacks:
>
> * [user managed [...]

MKS-TFT is now inline with BTT master. remote firmware is implemented in the wifi  side+darkspr1te bootloader.

Andrei-Pozolotin, 2020-03-09:
great, thank you