makerbase-mks/MKS-TFT #286: MKS GEN L display not working

BentWerx, 2019-03-06:
I have a New MKS Gen-L V1.0 3D Printer Controller Board Relacement Ramps 1.4 Mega 2560 R3 and a KINGPRINT 12864 LCD Graphic Smart Display Controller Board with Adapter and Cable for 3D Printer Ramps 1.4 RepRap 3D Printer Mendel Prusa Arduino. 
On the second motherboard and still can not get the display to work. Using the newest marlin firmware and Arduino IDE to flash the board. You can hear the fan turn on after power up but no display.
Any help is appreciated

BentWerx, 2019-03-07:
[Configuration_adv.zip](https://github.com/makerbase-mks/MKS-TFT/files/2943408/Configuration_adv.zip)

makerbase-mks, 2019-03-11:
If you refreshed the firmware of gen-l, but lcd is not displayed. Maybe the pin of the screen is just right. Because different manufacturers may set the direction differently. You can try and connect the original cable by 180 degrees. So you may need to deal with the motherboard socket or the cable connector (because the motherboard socket has a foolproof opening)

BentWerx, 2019-03-11:
I tried that the screen just glows. I must not have the code right for the display. Thank you very much for responding.

tanowano, 2019-03-11:
Hi, yesterday I bought an MKS TFT 32 and connected it to my MKS GEN L 1.0, there is no communication even though the baud is correct, what can I do that is wrong, and update the TFT 32 to the latest version. Sorry for my English

Stady234, 2019-04-07:
can you attach a picture of the cable connected to the board and screen?  I might be able to help...

makerbase-mks, 2019-04-07:
Yes, can you attach the connection diagram of your motherboard and screen (back)?
Check if there is a problem with the wiring

closed 2020-06-06 by makerbase-mks