makerbase-mks/MKS-TFT #288: Mesh Bed Leveling?

Oxika95, 2019-03-12:
I see the options for bed leveling in the config file here...

#leveling mode(manual:0; auto:1; conceal leveling button:2)
>cfg_leveling_mode:0

but it only shows manual auto and conceal. I don't have a probe yet but I was hoping to use manual mesh bed leveling as shown here

https://www.youtube.com/watch?v=X6lnlc-Vu-A

has anyone gotten this to work?

makerbase-mks, 2019-03-15:
In the tft configuration, only the configuration interface displays whether it is automatic or manual.
The collective configuration still needs to be configured on the motherboard.

Oxika95, 2019-03-15:
Thank you for the response I will give that a shot however, could you take a look at my reply to this thread...

https://github.com/makerbase-mks/MKS-TFT/issues/273

As the issue I am having here is preventing me from using the board at all.

mks-viva, 2021-02-24:
Has this issues been resolved?

Oxika95, 2021-02-24:
Talk about a nekro, this was an issue back in 2019 and this is 2021. We got a whole pandemic in between now and then. 

No I don't think this issue ever got resolved I had to stop using the board. You can close this thread though.

closed 2021-02-25 by mks-viva