makerbase-mks/MKS-TFT #290: Connection MKS GEN L V1.0 et MKS TFT 35 OU TFT 24

domichele, 2019-03-16:
Bonsoir
J ai un problème depuis des mois
pour moi il est impossible de faire connecte mes écrans MKS TFT 35 ou 24 avec ma carte MKS GEN L V 1.0 la vitesse est a 115200 partout 
mon firmware marlin 1.1.8 est vérifier enregistre et téléverser
le port est bon
Je ne peux pas non plus faire communiquée repetier avec la carte Gen 
La seul chose que je peux faire sais déplacer les axes et chauffer avec mon écran lcd
Si quel qu un peut me de-paner MERCI

makerbase-mks, 2019-03-21:
If it is not communication, it should be the problem of baud rate.
The default baud rate of the touch screen is 250,000

If you are not sure if the motherboard and screen baud rates are the same, reconfirm the settings and refresh. If you need the touch screen firmware you can go to this website to download. You can edit the baud rate in mks-config, save it to the sd card, and insert it to the screen to update the configuration.

domichele, 2019-03-21:
Bonsoir

Merci de répondre

Je suis loin d être bon en informatique 

 

J ai péniblement installer une nouvelle carte sur mon imprimante

D ANS MARLIN 1.1.8

La vitesse est de 115200 ,dans la configu txt de l’écran j ai modifier  1 pour marlin et 3 pour 115200

Sais tout ce que j ai changer 

Le deuxième problème qui pour moi est lier est que je ne peux pas connecter repetier a la carte de l’imprimante

Merci

 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

makerbase-mks, 2020-03-12:
The motherboard and the touch screen cannot communicate, which is usually caused by the baud rate not being maintained.

closed 2020-03-12 by makerbase-mks