makerbase-mks/MKS-TFT #291: firmware for mks v4.0 TFT

adarshparavoor, 2019-03-17:
Where can i get the latest firmware for TFT V 4.0, which not available. Please provide.

IrishMoss, 2019-04-25:
i have a v4.0 display and am using the latest 3.0.3 firmware, and it works fine.
i don't think the hardware version numbers coincide with the firmware versions.

labalast, 2019-04-29:
I am trying to update firmware 3.0.2 to 3.0.3 of a TFT28 V3.0, and following the instructions of innumerous sites on how to do it, still is not working. I Have copied all files from 3.0.3 release to the root directory of a 512MB SD card, inserted into TFT28 slot, and turned on. Does nothing, and About still shows 3.0.2.
Can IrishMoss  help?

IrishMoss, 2019-04-29:
No idea why the firmware wouldn't load from your SD card.
Maybe you're missing some of the files?
Here's a video I followed for updating the firmware: https://www.youtube.com/watch?v=ncc7TRkCH-M

labalast, 2019-04-29:
Thank you IrishMoss for your reply. I had seen that video and followed those instructions. I had a RepRap firmware which came with the TFT and replaced it with the 3.0.2 firmware from 3D MS. Since it could not read any gcode file from the SD card , I am trying to replace with 3.0.3. I copied all the files (mks_pic, mks_config.txt, mkstft28.bin etc)from the classic example to a formatted 512MB SD as said, and inserted it in the TFT28 slot. 
No luck, the About function still shows 3.02. TFT28 selected to run on 5V,  is connected to a MKS RUMBA+ Board running Repetier Firmware, and mks_config.txt was changed accordingly.
I am running out of options, and would appreciate any help.

labalast, 2019-04-29:
Thank you for your reply. Have commented more on GitHub.

IrishMoss, 2019-04-29:
It shouldn't really matter what it's connected to.  You can update the firmware with the tft just powered over usb and not connected to anything else.
It should be as simple as copy appropriate files to SD, insert SD to TFT, power up TFT.
If it doesn't automatically start flashing when powering up, then either files are missing (or corrupt), or there's something wrong with the TFT device itself.
Maybe try a different SD card.

labalast, 2019-04-30:
Thanks again,
Never tried updating powered over usb, always used the cable from board´s
AUX1 to TFT.
Will try doing this and using a different SD card.

labalast, 2019-04-30:
No luck!
Tried new SD card,and booting only powering through 12V dc input (don´t
have right usb cable to connect directly to TFT usb), using new files from
3.0.3 release, Classic_En_Blue. Tft shows booting..., but maintains 3.0.2
version.
Probably something wrong with TFT board.

kleverline, 2019-05-09:
Make sure you go into the file system menu and set the source as SD card.  I had a similar issue and that fixed it.  A few times I also had to do the following (after setting the file system source)

1. Add update files to SD and insert in TFT
2. Power on TFT (no update happens)
3. Go into file system menu and ensure it is set to SD card
4.*Main Point*  Use the phycial reset button to reboot the TFT.  Here is the important part, you can't just tap the reset button or it will just reboot without triggering the update (same with power cycling via unplugging the power).  I had to hold the reset button for a second or two (basically just a "long press") and then when I let go it booted into the firmware update.

labalast, 2019-05-09:
Thank you kleverline for your reply.
I had done all these steps, also without any luck. Finally I decided that
the bootloader was corrupted, and using STLink V2 I erased the TFT28 chip,
which now is a blank chip. Now I am looking for an alternative bootloader
to upload to the chip. This far, the ones available in the web (hackaday,
github etc) also did not work. I hope someone could help on that.

darkspr1te, 2019-08-08:
For those that have erased the device and have V1 or v4 hardware see my repo for factory bootloaders to restore your device,https://github.com/darkspr1te/MKS-TFT-V4-alt-fw.git
In time i'll post a opensource bootloader that will support the factory bin files. 
Enjoy :-)

delwinbest, 2019-10-11:
> For those that have erased the device and have V1 or v4 hardware see my repo for factory bootloaders to restore your device,https://github.com/darkspr1te/MKS-TFT-V4-alt-fw.git
> In time i'll post a opensource bootloader that will support the factory bin files.
> Enjoy :-)

Howdy! Thanks for sharing. I note your readme stats that "This is not firmware to provide original functionality". By 'restore' are you saying this does now support original FW upload and run? 

If yes, HoW on earth did you get the FW?

darkspr1te, 2019-10-11:
There is the original bootloader there which can then run the original (MKS provided) firmware. My firmware will run on the factory bootloader or without and is currently only a Proof Of Concept which is what the readme refers too. As to how I got a copy is easy, i wrote firmware to dump the flash to sd-card. I then wrote my own bootloader and firmware for my purpose , both of which will be in the arduino environment soon (via a json URL). My intention is not to use this board on a printer so I never intended to 'need'the factory bootloader but after seeing some users who had managed to erase it I figured I would add it here. I will update the readme to clarify this.

delwinbest, 2019-10-11:
I'm going to test this evening on my dev MKS board and may nominate you for 'You're a Legend' award. Fantastic. I've also started building out base FW for the v4.0 board seeing as it was just lying around, bricked.

darkspr1te, 2019-10-11:
Feel free to fork my repo, I currently have everything working at a base level, eg all uarts, spi, lcd & gpio.  USB Host not yet, but stm(stm32duino repo, https://github.com/stm32duino/Arduino_Core_STM32 ) have my commits on this and should be live soon. the gfx part is a modded lib from lady ada which i need to fork and incorporate correctly , my code is a serious mess of patches as the lcd is 9-bit not spi , which is why I chose it, when in 9 bit mode you can trest the lcd as ram (even thought the stm chip does not support this feature at hardware level) you can still create a basic frame buffer.

darkspr1te, 2019-10-11:
:-/ , I've just noticed that i've not updated my online git with my offline so only the factory bin files are showing, i will upload the commit with working code for anyone wishing to play with it tonight.

delwinbest, 2019-10-11:
I was a little confused :) when i had a poke. :) I did not realize that you could read from flash within code. I assumed code protect of some type would be enabled. So you uploaded your custom bin at the flash offset location and read back the bootloader?

darkspr1te, 2019-10-11:
> I was a little confused :) when i had a poke. :) I did not realize that you could read from flash within code. I assumed code protect of some type would be enabled. So you uploaded your custom bin at the flash offset location and read back the bootloader?

yes, that exactly it. There's no point having CROP on as the actual firmware is available and not encrypted (like most OBD readers based on stm chips) so anyone could reverse engineer the firmware but it would be a waste of time and quicker to write new code. in the case of the bootloader itself 90% of it is stm factory booloader sample with lcd code added. so no point reversing that either. Also to clarify you cant read flash from ram code only rom code (so if code resides in 0x00000 then flash read possible but if code is 0x800000 then you cant read flash) so when the bootloader writes my code to the flash it's now at 0x7000 and has access to flash.

delwinbest, 2019-10-11:
@darkspr1te you are officially a Legend. Your bootloader dump restored my MKS TFT28 v4.0 board. Looks like the dump is from a larger screen type, but that s ok. 
If you like, mind sharing the code you used to copy the bootloader to SD? I have a new TFT28 board arriving soon and I'll grab the bootloader form that unit with the smaller screen.  When I get some time I'll put up a 'how to restore' guide to the folks out there.

darkspr1te, 2019-10-15:
I have added the dumpers source code to the repo. I will be adding a open source bootloader compatible with MKS firmware on v4 boards, I had written this prior to getting the backup as it was never my intention to use the factory stuff but it was a good exercise anyway and you could add features if you wish or fix bugs (like USB host mode for updating firmware does not work on my fw)

wilensky, 2020-04-20:
So I faced several problems (_booltloop, `booting ...` screen_) on updating my _MKS TFT32_L v4.0_ with latest firmware available.

And what I found out is:

1. Download latest firmware and extract `Example` folder of your kind to your SD/USB.
1. Go to **`mks_config.txt`** and set **`>cfg_baud_rate:3`** which is equal to **`115200`**
1. Flash firmware. This should get screen flashed but printer unresponsive.
1. **Again!** Copy `mks_config.txt` from archive and **set `>cfg_baud_rate` to your value** (_`4` in my case_)
1. Run TFT with updated `mks_config.txt`
1. Profit ... seems to be the hardest word ...

Hope this guide will help someone to get around this problem.

> P.S: My main board is **MKS Gen V1.4**

Kodbee, 2020-05-17:
Good afternoon,
I have the same problem. My MKS TFT28 4.0 does not communicate with Ramps 1.4. I have updated the firmwares of the MKS TFT28 4.0 and Marlin several times, and I have verified that both are at 250000 bauds, but it is impossible communicate. However, when I connect to the computer, I can actually move the motors with the Repetier Host, but I can´t with the MKS. I suspect the problem may be version 4.0. Does anyone have a solution? Does anyone manage to communicate the Ramps with MKS TFT28 version 4.0?
Regards,

darkspr1te, 2020-05-18:
> Good afternoon,
> I have the same problem. My MKS TFT28 4.0 does not communicate with Ramps 1.4. I have updated the firmwares of the MKS TFT28 4.0 and Marlin several times, and I have verified that [...]

I have had this same issue with a ramps board and it cannot be fixed in software. This is due to the pull up/pull behavior of the Arduino ramps boards. When i connect to my ramps and view on a scope you can clearly see the RX of the ramps board does not drop below 3.9v and the threshold for RX to see a 0 instead of a 1 is 2.5v, i did however manage to get it to function by using a level converter I got from adafruit. Maybe the same will work for you. 
darkspr1te

darkspr1te, 2020-05-18:
Correction, I have fixed this issue in software, see my BTT firmware repo for alternative firmware that "may" fix your issue. Backups are available for bootloader/mks firmware& eeproms to restore function so long as you have access to a serial adapter or better a stlink adapter.

Kodbee, 2020-05-19:
Thanks! but I have not access to a serial adapter or a stlink adapter... There are people that use MKS TFT28 V4 and Ramps 1.4 without problems... there must be an easier way to make them work together...

mshavandi, 2020-10-03:
guys help please! The mkstft70 firmware is not uploading and updating the chip. I'm using USB for uploading the files, do you think it's the problem?? I have tried all solutions provided here but no luck. I can move my printer axes easily with the LCD but cannot change configuration and upload my custom firmware to it (or even change the theme). Please give me a solution. thanks

darkspr1te, 2020-10-03:
as far as i am aware the bootloaders only update from sd-card , if USB-Host is included in a build the code size jumps quite a lot making the bootloader large, also some mks models use the MKS-Robin encrypt system so any custom firmware 'may' require that depending on model, i have the mkstft35 and the build environment is the same, that only updates via sd-card and requires mks-robin encrypt or the firmware will not boot correctly. 
see https://github.com/darkspr1te/mkstft35_V1_open_bootloader/blob/master/Src/flash.c for the encrypt key and source  code to encrypt/decrypt if you are building custom firmware, somewhere i've also published a python script foe the same