makerbase-mks/MKS-TFT #292: TFT35 Status Screen

furymountain, 2019-03-17:
Hi,

There seems to be a status screen for the smaller TFT, but not for TFT35. It would be great to have it when not printing directly from the TFT to see temperatures, fans, maybe even layerstatus and percentage if possible as a default screen.

Also a Z offset button when printing would be great. Right now i can only find it when not printing.

Best regards
Andreas

makerbase-mks, 2019-03-21:
Are you talking about tft35, or other screens?

furymountain, 2019-03-21:
I have the TFT35 and there doesn't seem to be any status screen where I can see temperature of bed/nozzle and so on at the same time. I have to go in to each menu to see it. It would be nice if there was a genereal status screen as the default screen.

behold81, 2019-05-04:
I just updated min from 3.0.1 to 3.0.3 and the status screen has gon from my 3.0 screen

behold81, 2019-05-04:
solved. use the Retro version

closed 2019-05-06 by makerbase-mks

furymountain, 2019-06-05:
I believe you are talking about TFT28 and not TFT35. The firmware for TFT35 does not have a retro version. There is only one.

thonumar, 2020-02-15:
pessoal descobri que tem sim a versão retro para mks tft 35 mas no arquivo de configuração não mostra então tentei varia maneiras até conseguir e com muita insistência descobri uma forma de funcionar retro no tft 35 vou ensinar para vocês abra o arquivo  robin_nano35_cfg dentro dele você vera uma linha assim: >cfg_screen_display_mode	0		#the style of display on the screen home(0:classic,1:simple)

para quem não estiver encontrando procure no campo de busca control + F

então eu acrescentei o numero 2 e funciona como retro ficando assim:
mas o problema maior é que os 3 ícones: ( temperatura extrusor 1 ) ( mesa ) ( ventilador ) não estou conseguindo editar a imagem para colocar neles, agora ajudei vocês kkkkk preciso de ajuda também kkkkk preciso que todos nos nos unirmos para descobrir qual nome do arquivo para colocar na imagem do ícone assim posso colocar a imagem em meus painéis, aguardo resposta de voces.