makerbase-mks/MKS-TFT #293: TFT35 Simplify3d

furymountain, 2019-03-17:
There is a plugin for cura but none for Simplify3d which is my and probably a lot more peoples preference. Would it be possible to make a plugin for Simplify3d as well? In the meantime are there any good way to upload files to the SD card without the cura plugin? (since that can only upload flies from cura directly). I mean so that I can slice in any software and then upload the file via WiFi to the MKSTFT SD card and make it print from it, so I can se the status and everything just like if i manually copy to the SD card put it in the TFT and start the print from the screen?

There seems to be an upload to SD card with pronterface, but it doesn't actually upload the file it just starts the print over WiFi and no status is displayed on the screen.

Best regards
Andreas

makerbase-mks, 2019-03-21:
We and our R&D department give feedback to see if it is adjusted later.

Wranglers, 2019-06-13:
Is there any news on the simplefy3d plugin for de gcode preview on the mks tft35.
Thx Erik

mcsultan, 2019-11-16:
i need too.

Cemileri, 2020-09-01:
I think all mks users will need it..

doug-scott, 2020-09-21:
Maybe not all mks users, but all SF3D users will appreciate it.  Cura may be an easy slicer to use, if you have been using it solely for years, but, SF3D is not a free slicer, and once you purchase it you tend to want to stay with what works for you.  That being said, Simplify3D has been hinting at a version 5 of the software for a couple of years now.  They now completely ignore requests for updates on the release schedule.  Perhaps MKS will want to wait and see if Simplify3D is even around at year end.  MKS cannot create the internal links that SF3D needs to support without co-operation from SF3D.

Scottone27, 2023-08-03:
I need this as well, i regret purchasing S3dV5, seems useless so far