makerbase-mks/MKS-TFT #294: MKS-TFT70 "turn off motor" not working

rafaelferrari0, 2019-03-20:


rafaelferrari0, 2019-03-20:
Hi
I've updated my TFT70 to version MKS-TFT70 V1.0.2
Supposedly this version fixes the bug about turning off motors, but this not work.

Sniffing serial communication from TFT to MKS SBASE (smoothieware), the button should send 'M84', but only sends garbage over serial interface.
 [image]

Others buttons works normally, sends corresponding Gcode.

Please, fix the bugs or open the source code if you are not able to work. #285
Do you ever test your firmware, or monitor serial communication ?
 
This is a expensive display, not work as expected. Makerbase is becoming a bad company with poor support.

makerbase-mks, 2019-03-21:
Hello there
On our side, it is normal to test this function.
You can't use it normally, you can provide the firmware and configuration files of tft70 and sbase that can't normally shut down the motor now.
Let's test it here.

rafaelferrari0, 2019-03-25:
> Hello there
> On our side, it is normal to test this function.
> You can't use it normally, you can provide the firmware and configuration files of tft70 and sbase that can't normally shut down the motor now.
> Let's test it here.

Hi, here is my TFT70 configuration file: (rename to mks_config.txt)
https://raw.githubusercontent.com/rafaelferrari0/trash/master/mks_config_tft70.txt

The firmware is latest version (1.0.2):
https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS-TFT7.0/TFT70_V1.0.2%20Release%20file.rar

MKS SBASE is using latest stable: (because we need some new features)
https://github.com/Smoothieware/Smoothieware/blob/master/FirmwareBin/firmware.bin
Smothieware configuration is defaults, with some adjustments:
uart0.baud_rate                              57600
dfu_enable                                  true

This setup is working correctly, and MKS SBASE responds normally to M84 commands sent by Repetier-Host.

mks-viva, 2021-02-24:
Hello! You can update TFT firmware and test it on your board