makerbase-mks/MKS-TFT #298: Creating Cusotm Button issues

PhillyNJ, 2019-03-24:
I am trying to create a custom button ( bmp_custom1.bin), I am using the program Image2lcs to convert the 16 bit bmp to a .bin file. When I try to flash the MKS, it hangs on the bmp_custom1.bin. 

Here is a screenshot of my issue and settings. Thanks for the help.

https://i.imgur.com/gqCjU9X.jpg

PhillyNJ, 2019-03-26:
{{Bump}} - Does anyone monitor this page?

IrishMoss, 2019-04-25:
Try the makerbase image editor to convert.  That one has worked well for me.
https://github.com/makerbase-mks/Software/tree/master/MKS%20TFT%20image%20convert

kleverline, 2019-05-09:
> {{Bump}} - Does anyone monitor this page?

Not the devs, it seems.

PhillyNJ, 2019-05-09:
I don’t think so. I posted a question months ago and no one responded. 

Regards,

Phil

Sent from Phil's iPhone