makerbase-mks/MKS-TFT #300: How do I set the pre-heat temperatures under the "heat" button?

josef, 2019-03-27:
Hi, I would like to have the temperatures pre-set to 200 for the extruder and 60 for the bed when hitting the "heat" button on the MKS v3.0.3 TFT how is this possible?

[image]
[image]
[image]

FranckGib, 2019-03-31:
Hello,
you have to create a button and put the corresponding GCODE:
M104 S200; M140 S60;

makerbase-mks, 2019-04-08:
You can customize the button instructions through the configuration file.

############## Function Customization #######################################

#user-defined function1 and 2.(display this button .  disable: 0; enable: 1)
>cfg_function_btn1_display:1

#command of user-defined function
>function_btn1_cmd:M84;

#set number of  "More" button 
>moreitem_pic_cnt:0

#edit command for 1~7 "More" button,each command must be separated by semicolon ";"
>moreitem_button1_cmd:G28 X0;

>moreitem_button2_cmd:G28 X0;

>moreitem_button3_cmd:G28 Y0;

>moreitem_button4_cmd:G28 Y0;

>moreitem_button5_cmd:G28 Z0;

>moreitem_button6_cmd:G28 Z0;

>moreitem_button7_cmd:G28;

#how many "More" button display on screen interface
>morefunc_cnt:0

#command setting on button1~6
>morefunc1_cmd:G28;
>morefunc2_cmd:G28;
>morefunc3_cmd:G28;
>morefunc4_cmd:G28;
>morefunc5_cmd:G28;
>morefunc6_cmd:G28;

#---------------------------------------------------------------------------------

closed 2020-03-12 by makerbase-mks