makerbase-mks/MKS-TFT #302: MKS TFT35 interface

tomcatmobile, 2019-04-05:
Thank you very match for you work, but the interface of the device is very inconvenient even compared to the 12864 from Marlin. This needs to be finalized.
If you wish, I can prepare sketches for the interface, with a convenient arrangement of buttons and information text or, if you tell me where to find the source code, I could write a part of the interface for this device

JDSymington, 2019-04-05:
I just got mine and I fully agree with you. I'm considering just buying a 12864 as the functionality on the TFT is really limited.

tomcatmobile, 2019-04-05:
I took the TFT to replace the 12860. but now I have both displays connected. because it is impossible to use only TFT

JDSymington, 2019-04-05:
If I may ask, why did you decide to replace your 12864?

tomcatmobile, 2019-04-05:
I wanted to use usb flash and wi-fi. I was also interested in the touch screen. I did not know that after this I would get an almost unusable device

JDSymington, 2019-04-05:
Octoprint takes care of the WiFi for me. 
The touchscreen is a nice to have but not worth it for all the features you loose.
The biggest downside for me it that I can't change EEPROM settings using the TFT.

tomcatmobile, 2019-04-05:
Complete agreement. Now I think I was wrong. I hope the developers will hear us otherwise I will try to sell this display and buy a bigtreetech

odersone, 2019-05-15:
+1 on these issues, I just cannot recomend MKS products based on mye experiences. Seeing the Duet maestro retail at 90£, I regret getting Gen L + TFT35.

JDSymington, 2019-06-27:
I sold my MKS TFT35 and bought a normal RepRapDiscount Smart Controller. 
It is cheaper and it actually works. The SD-Card reader is also working so I'm not looking back.

I definitely can't recommend the MKS TFT35!

carl1961, 2019-06-29:
BigTreeTech with open source will put a big hurt on these closed source suppliers, being greedy and trying to hold the market has put the hurt on many a company , you would think companies would learn from the past. All my MKS parts, or outdated or useless as the post above share.

tomcatmobile, 2019-06-29:
therefore, I offered the developers help in the interface project with the necessary commands without getting the source code from them

carl1961, 2019-06-29:
tomcatmobile, is it possible you could do work with the BigTreeTech 3.5 TFT source files to add custom buttons and menu to do more?

carl1961, 2019-06-29:
tomcatmobile, is it possible you could do work with the BigTreeTech 3.5 TFT source files to add custom buttons and menu to do more? It is also a serial TFT, like the MKS

fnhhnxfn, 2020-02-02:
i did it to mine and i lost my display , copied BTT screen firmware.bin file and just updated the firmware with it and it updated sucessfully after updating my screen turned black and i couldnt turn it on , is there any ideas guys? i really need help plzzzz help me

betocabus, 2020-02-07:
> i did it to mine and i lost my display , copied BTT screen firmware.bin file and just updated the firmware with it and it updated sucessfully after updating my screen turned black and i couldnt turn it on , is there any ideas guys? i really need help plzzzz help me

after a hole day, i pressed the SD with a little force inside the sd reader and HELD it! the update firmware started (still Holding, hurting my finger) and only worked with this method, very weird (i have 2 btt screens(tft 35 v2, same problem in both) with this i could update and put it to work again.