makerbase-mks/MKS-TFT #303: Position is interpretet as bedtemperatur

dudi300, 2019-04-06:
I use Marlin in my 3D printer as FW, now 2.0 on a SBase 1.3 together with a TFT3.2 / Classic FW.  So during printing I can see the temperature. Sometimes, the shown bedtemperatur is wrong. 
I have now dumped the serial data stream and found out, that positions are misinterpreted as bedtemperature.

Count A: 53662 **B: 221** Z: 789B: 49.75 / 50.00.ok. <--- That's finished
**B: 49.75** / 50.00 @: 93 B @: 127.ok. <--- And that's also complete.

It seems to me, that the FW just looking for the token "B:" and interpret this as bed temperature.

Here is the link with the datatdump:https://github.com/MarlinFirmware/Marlin/issues/10735

Best Regards

makerbase-mks, 2019-04-07:
Is the temperature error occurring occasionally or always occurring?
Because the screen acquires the temperature, but is obtained by transmitting the instruction of M105. If the display temperature is wrong, it may be a firmware problem or it may be the sensor or the temperature measurement port of the motherboard. But if it doesn't always happen, it shouldn't be a hardware problem.

dudi300, 2019-04-12:
Hi,
Now I have a little time to answer. I hope it helps....

This is the traffic on the Rx-Pin of the  TFT.

X:147.68 Y:150.51 Z:2.90 E:0.00 Count A:46917 B:3861 Z:2281
X:147.68 Y:189.00 Z:2.90 E:0.00 Count A:47582 B:3032 Z:2287
X:130.74 Y:197.00 Z:2.90 E:0.00 Count A:54729 B:-7777 Z:2325
X:143.60 Y:193.00 Z:2.90 E:0.00 Count A:55030 B:-8010 Z:2325
**ok T:196.82 /195.00 B:50.20 /50.00 @:67 B@:0**
X:131.24 Y:150.33 Z:2.90 E:0.00 Count A:46323 B:-4335 Z:2292
X:129.44 Y:146.92 Z:3.00 E:0.00 Count A:47055 B:-1051 Z:2310
**ok T:195.00 /195.00 B:50.20 /50.00 @:86 B@:0**
X:129.65 Y:147.26 Z:3.00 E:0.00 Count A:47353 B:-287 Z:2394
X:128.44 Y:146.41 Z:3.00 E:0.00 Count A:47284 B:-356 Z:2394
X:128.12 Y:146.18 Z:3.00 E:0.00 Count A:52441 B:-10873 Z:2369
X:129.45 Y:147.95 Z:3.00 E:0.00 Count A:52470 B:-10959 Z:2368
X:129.74 Y:148.24 Z:3.00 E:0.00 Count A:47184 B:-418 Z:2393
X:129.19 Y:147.07 Z:3.00 E:0.00 Count A:47147 B:-502 Z:2392
X:127.88 Y:149.51 Z:3.00 E:0.00 Count A:46712 B:-522 Z:2388
X:128.39 Y:194.82 Z:3.00 E:0.00 Count A:50904 B:-9906 Z:2370
X:175.95 Y:191.39 Z:3.00 E:0.00 Count A:51895 B:-10713 Z:2368
X:183.44 Y:140.34 Z:3.00 E:0.00 Count A:60769 B:-1333 Z:2343
X:175.60 Y:133.18 Z:3.00 E:0.00 Count A:50081 B:6075 Z:2334
X:186.12 Y:135.40 Z:3.00 E:0.00 Count A:51402 B:8146 Z:2318
X:185.28 Y:133.00 Z:3.00 E:0.00 Count A:51060 B:8602 Z:2316
X:166.21 Y:132.68 Z:3.00 E:0.00 Count A:53227 B:10638 Z:2359
X:160.11 Y:133.00 Z:3.00 E:0.00 Count A:51846 B:7385 Z:2318
**ok T:194.55 /195.00 B:49.75 /50.00 @:87 B@:127**

The following stream will be automatically sent by Marlin 2.0 without any request from the TFT (mesured on the open Aux1 line).

B: is shown on the display as a temperature. It only happens during printing, only then the string be sent.

X:186.01 Y:197.32 Z:1.90 E:0.00 Count A:62775 **B:1647** Z:1514
X:201.32 Y:193.73 Z:1.90 E:0.00 Count A:61248 B:1256 Z:1492
X:174.18 Y:191.15 Z:1.90 E:0.00 Count A:60574 B:-2922 Z:1461
X:132.68 Y:197.32 Z:1.90 E:0.00 Count A:54722 B:-8104 Z:1523
X:147.68 Y:189.00 Z:1.90 E:0.00 Count A:58230 B:-4907 Z:1491
X:130.74 Y:197.00 Z:1.90 E:0.00 Count A:54729 B:-7777 Z:1526
X:143.60 Y:193.00 Z:1.90 E:0.00 Count A:54606 B:-8434 Z:1519
X:131.21 Y:158.33 Z:1.90 E:0.00 Count A:49837 B:-2109 Z:1523
X:143.76 Y:150.33 Z:1.90 E:0.00 Count A:49310 B:-2210 Z:1520
X:179.35 Y:134.33 Z:1.90 E:0.00 Count A:51087 B:7222 Z:1429
X:169.05 Y:185.00 Z:1.90 E:0.00 Count A:50270 B:6856 Z:1435
X:181.65 Y:195.67 Z:1.90 E:0.00 Count A:59467 B:-2231 Z:1465
X:184.99 Y:197.41 Z:2.00 E:0.00 Count A:59749 B:-2586 Z:1465
X:176.01 Y:197.42 Z:2.00 E:0.00 Count A:63358 B:139 Z:1582
X:176.01 Y:197.42 Z:2.00 E:0.00 Count A:54372 B:-7204 Z:1608
X:156.01 Y:154.60 Z:2.00 E:0.00 Count A:58626 B:-2535 Z:1554
X:185.37 Y:138.84 Z:2.00 E:0.00 Count A:50515 B:8117 Z:1502
X:185.29 Y:138.16 Z:2.00 E:0.00 Count A:53596 B:10716 Z:1559
X:176.30 Y:139.14 Z:2.00 E:0.00 Count A:49781 B:7642 Z:1509
X:201.76 Y:180.39 Z:2.00 E:0.00 Count A:51642 B:7105 Z:1509
X:176.29 Y:190.88 Z:2.00 E:0.00 Count A:60129 B:-3143 Z:1546
X:176.73 Y:190.95 Z:2.00 E:0.00 Count A:54577 B:-8032 Z:1602
X:176.73 Y:190.95 Z:2.00 E:0.00 Count A:54493 B:-7890 Z:1603
X:183.95 Y:190.58 Z:2.00 E:0.00 Count A:54236 B:-7324 Z:1606
X:202.16 Y:150.97 Z:2.00 E:0.00 Count A:59924 B:-1057 Z:1535
X:176.86 Y:139.45 Z:2.00 E:0.00 Count A:49657 B:7329 Z:1512
X:177.56 Y:139.34 Z:2.00 E:0.00 Count A:49529 B:7409 Z:1512
X:176.19 Y:132.31 Z:2.00 E:0.00 Count A:49948 B:6391 Z:1522
X:177.10 Y:131.98 Z:2.00 E:0.00 Count A:50389 B:6146 Z:1523
X:178.45 Y:131.75 Z:2.00 E:0.00 Count A:50509 B:6124 Z:1523
X:182.06 Y:131.54 Z:2.00 E:0.00 Count A:50034 B:6320 Z:1522
X:183.79 Y:132.07 Z:2.00 E:0.00 Count A:49360 B:7025 Z:1516
X:184.64 Y:132.35 Z:2.00 E:0.00 Count A:49451 B:7229 Z:1514
X:201.52 Y:148.85 Z:2.00 E:0.00 Count A:49633 B:7480 Z:1511
X:202.12 Y:149.95 Z:2.00 E:0.00 Count A:50179 B:8105 Z:1503
X:202.52 Y:151.24 Z:2.00 E:0.00 Count A:50538 B:8295 Z:1501
X:202.52 Y:178.40 Z:2.00 E:0.00 Count A:56059 B:8431 Z:1575
X:202.12 Y:179.93 Z:2.00 E:0.00 Count A:56147 B:8436 Z:1576
X:183.95 Y:198.10 Z:2.00 E:0.00 Count A:56602 B:8199 Z:1580
X:182.70 Y:198.21 Z:2.00 E:0.00 Count A:56844 B:8018 Z:1582
X:179.00 Y:198.52 Z:2.00 E:0.00 Count A:60946 B:3855 Z:1597
X:177.27 Y:197.98 Z:2.00 E:0.00 Count A:61127 B:1881 Z:1579
X:176.41 Y:197.71 Z:2.00 E:0.00 Count A:61126 B:-2272 Z:1534
X:128.89 Y:196.01 Z:2.00 E:0.00 Count A:60946 B:-2489 Z:1537
X:127.88 Y:193.63 Z:2.00 E:0.00 Count A:60037 B:-3358 Z:1549
X:127.88 Y:191.93 Z:2.00 E:0.00 Count A:51891 B:-10715 Z:1563
X:128.12 Y:180.95 Z:2.00 E:0.00 Count A:51442 B:-10514 Z:1563
X:128.51 Y:147.75 Z:2.00 E:0.00 Count A:44832 B:-3834 Z:1561
X:131.70 Y:145.84 Z:2.00 E:0.00 Count A:44654 B:-3726 Z:1560
X:174.95 Y:138.08 Z:2.00 E:0.00 Count A:47440 B:4982 Z:1543
X:147.68 Y:150.51 Z:2.00 E:0.00 Count A:44654 B:-1970 Z:1564
X:156.43 Y:153.00 Z:2.00 E:0.00 Count A:46996 B:-479 Z:1589
X:158.41 Y:134.82 Z:2.00 E:0.00 Count A:50031 B:5558 Z:1530
X:185.60 Y:133.00 Z:2.00 E:0.00 Count A:47825 B:5367 Z:1538
X:201.32 Y:145.65 Z:2.00 E:0.00 Count A:54135 B:10287 Z:1564
X:186.97 Y:138.89 Z:2.00 E:0.00 Count A:52800 B:10337 Z:1545
X:200.50 Y:183.00 Z:2.00 E:0.00 Count A:54914 B:9503 Z:1568
X:186.01 Y:197.32 Z:2.00 E:0.00 Count A:62775 B:1647 Z:1594

dudi300, 2019-06-26:
It only occurs during printing. 

The appendix shows the complete serial stream.

Everything that has been sent seems to me plausible.