makerbase-mks/MKS-TFT #304: Disable/enable comunication

davidcgu, 2019-04-07:
There is any command I could use to create a button to enable/disable serial communication between screen and the board?

From there side I remember I think there were only used 4 pins on the connection, 2 for power supply(+5Vdc-0) and 2 for serial (Rx-Tx)  can you be so kind to indicate me which ones for each function?

[image]

Thanks in advance

makerbase-mks, 2019-04-08:
5v and g are for power supply, TX and RX are for serial port transmission.
The screen should not be able to disconnect the serial port by instruction.

davidcgu, 2019-04-08:
Can you let me know the pinout on the connector?

makerbase-mks, 2020-06-06:
You can refer to MKS Hardware,like Gen_L and so on

closed 2020-06-06 by makerbase-mks