makerbase-mks/MKS-TFT #305: TFT 35 Display Issues and Corrupted .bin image files

kleverline, 2019-04-11:
I have a TFT 35 and the button icons are not displaying properly on the LCD.  The text under each button has a blue background instead of being white on black like it should be.  It looks like the text is highlighted.  I cannot find a way to adjust this in the config.  

Additionally, when I created my own boot logo it does not display right either.  I made the image 480x320 per the documentation and converted it to .bin with TFT Image Convert.  It displays normally in the TFT Image Convert program, but when I load it to the TFT35 the boot screen shows up distorted  and has a broken portion of the image multiplied across the top half of the screen.

Also, when I try to use the TFT Image Convert to view the .bin files from the firmware for the TFT35, they are all scrambled and do not look at all like they do on the display.  If I view images from the TFT24,28,etc. firmware files, they display normally.

There seem to be serious issues with the images on this display.  I have tried FW 1.0.1 and 1.0.2, downloaded clean copies multiple times, flashed them multiple times, and tried using TFT Image Convert on a different machine.  I always get the same errors.  

To add one more, if I turn off multiple languages in the config file, the TFT35 still displays the language change button and all 7 languages inside.

mooza3d, 2019-05-08:
I am having the same issues, especially the one with the logo. I can't get the bmp_logo.bin file to show properly on the tft35 display

kleverline, 2019-05-09:
Has MKS abandoned this repository?  4 weeks and no response at all?

tedbeer, 2020-01-06:
I created my custom logo successfully. When you convert your .BMP file into .BIN format specify to output 16 bit (rgb 565) and _NO_HEADER_ (uncheck the check box in the left panel)

iz3man, 2020-02-18:
Has anyone sorted out how to make your OWN ICONS? 
Which format do they need to be? It seems they are not interchangeable between tft35 and any other tftXX.

handsomedude, 2020-04-20:
I'm having issue trying to convert any image or icons.

Put custom button on for power on and off m commands.  Buttons work fine, just I cannot convert an image to a bin file. When I do the button looks all garbage.  Looks like my image half size vertically and third size horizontal repeated three times like a blur.

Same effect when I open mks tft image convert. .all icons look messed up.

Load the tft28 firmware folder and they show up fine.

Any program to use that's for the tft3.5

handsomedude, 2020-04-20:
Finally got it.  Downloaded the tft icons from the tft7.0 folder on github.
Checked icon size.  

Tft button uses 117 by 140 icons.
Set image2lcd as horizontal scan,16 bit true colour and output to 16 bit colour on lower tabs. 
Load image and save to mks_pic folder and flashed. 

All good now.