makerbase-mks/MKS-TFT #311: A couple of doubts before updating

beckmx, 2019-04-26:
I have an old replicator dual with sailfish 7.7, I was wondering if I can use this firmware to install it directly on the screen and it would be compatible, or this screen only communicates with normal gcode params?

meistertek, 2019-04-30:
I believe ther stock LCD on your replicator probably sends g-codes to the Mightyboard. Are you asking if a MKS TFT 28 or similar could replace the stock LCD on your Replicator Dual?  I think this might work/  Looking at the pinouts where the orginal LCD plugs in for the mightyboard Rev E.  The pin array is called "INTERFACE" and looking at it there are pins for SPI, which is what the MKS TFT touch screens use to communicate in g-code to various motherboards.  Hope this helps you out.

beckmx, 2019-04-30:
Did help, I was looking more in depth and saw that my printer has a dwin screen, could this be compatible with those displays?

beckmx, 2019-04-30:
btw i use the ctc bizer, replicator dual clone with only the display different

meistertek, 2019-05-06:
Most people ditch their mightyboard and go with a MKS Base or SBase motherboard.  If I were you, I'd simply try wiring up the MKS TFT 28 and seeing if it works to move steppers etc.  If it does work then you are on to something new that no one has done or at least documented before.  I may covert over 20 Flashforge Creator Pros and Makerbot Replicator 2 3D printers to MKS touchscreens and motherboards.  I would then sell all the working Mightyboards from them on e-bay.

folkertj, 2022-10-06:
Did you ever get this to work? I have a TFT and I would like to plug it in directly to the mightyboard. I'm not totally sure what the pins all do. If you could hook it up that would be awesome. I've done basically every upgrade except this. Thanks.