makerbase-mks/MKS-TFT #312: TFT28, M112 Do not stop printer

Tomaszm700, 2019-04-28:
I have provided one of additionall buttons as EMG STOP with command M112. I have ABL Active as well. If I sent Home G28 or ABL G29 and would like to stop printer durring this operations with M112 it not working to the moment of procedure finish. M112 sent from paralleal connected PC program will stop printer immidietly. 

**Firmware:** Marlin 1.1.9
**System Board:** MKS Gen L
**LCD:** TFT 28

**Current behevior:** Command M112 executed in queue 
**Expected behevior:** Command M112 executed immidietly to stop the printer in EMG case.

IrishMoss, 2019-04-29:
I had similar queuing issues with temp commands, and got around it by sending a M0 S1 cmd to 'break' the queue, and then issue the next command.

Tomaszm700, 2019-05-01:
Thanks, I will try this